sakaki-/gentoo-on-rpi-64bit #99: Only 3.7 GIB RAM on RPI4B RAM 4G

ykla, 2019-08-22:
Hi, I using your pi64 latest system, but only 3.7 RAM I can use, Does it used by GPU?
 pi64 ~ # free -h
              total        used        free      shared  buff/cache   available
Mem:          3.7Gi        95Mi       3.3Gi       0.0Ki       270Mi       3.5Gi
Swap:         1.0Gi          0B       1.0Gi
pi64 ~ #

sakaki-, 2019-08-23:
Amongst other things, yes. Try running:
```console
demouser@pi64 ~ $ dmesg | grep Memory
```

ykla, 2019-08-26:
`pi64 ~ # dmesg | grep Memory
[    0.000000] Memory: 3581188K/3997696K available (9660K kernel code, 1132K rwdata, 2924K rodata, 1088K init, 935K bss, 154364K reserved, 262144K cma-reserved)
[  242.726493] usb 2-1: Product: TransMemory
[  243.746002] scsi 0:0:0:0: Direct-Access     TOSHIBA  TransMemory      PMAP PQ: 0 ANSI: 6
`

Askmewho, 2019-09-05:
thats expected.. set gpu_mem to 16 ... the cma allocates memory for the gpu on mesa driver.

closed 2019-11-10 by sakaki-