sakaki-/gentoo-on-rpi-64bit #101: Audio does not work on RPI 3 B+ (Jack 3.5)

NguyenTrongThinh, 2019-08-24:
I use bcmrpi3_defconfig to rebuild kernel version 4.19. I get kernel source at https://github.com/raspberrypi/linux

When I run command aplay -l. It show strange output. There is no HDMI cable connected to raspberry. 

$ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: vc4hdmi [vc4-hdmi], device 0: MAI PCM vc4-hdmi-hifi-0 []
Subdevices: 1/1
Subdevice #0: subdevice #0

And Audio on 3.5 Jack not working too.

sakaki-, 2019-08-24:
Hi @NguyenTrongThinh 

What do you have in `/boot/config.txt`?

Does it work for you if you use one of my binary kernels ([bcmrpi3-kernel](https://github.com/sakaki-/bcmrpi3-kernel) or [bcmrpi3-kernel-bis](https://github.com/sakaki-/bcmrpi3-kernel-bis))?

Also see my note [here](https://github.com/raspberrypi/linux/issues/3181#issuecomment-524561499).

Best, sakaki

NguyenTrongThinh, 2019-08-25:
Thank! change to vc4-fkms-v3d works for me but I still got strange output when run command aplay -l

aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: ALSA [bcm2835 ALSA], device 0: bcm2835 ALSA [bcm2835 ALSA]
  Subdevices: 7/7
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
card 0: ALSA [bcm2835 ALSA], device 1: bcm2835 IEC958/HDMI [bcm2835 IEC958/HDMI]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: ALSA [bcm2835 ALSA], device 2: bcm2835 IEC958/HDMI1 [bcm2835 IEC958/HDMI1]
  Subdevices: 1/1
  Subdevice #0: subdevice #0

So we move to [raspberry-linux](https://github.com/raspberrypi/linux/issues/3181#issuecomment-524561499) to continue our discussion.

closed 2019-08-25 by NguyenTrongThinh