sakaki-/gentoo-on-rpi-64bit #102: Q Is it possible to migrate to RPI4?

skulko, 2019-08-29:
Hello 
is there a way to migrate an Image running under rpi3 to rpi4? 

I read in RPi4 Compatibility? #91 that your image is not compatible per se but that the cpu is backwards compatible... so it should be possible to replace the firmware/dtbs files and boot it on rpi4, no?

thank you very much for your projects btw. sorry if this has already beens answered..
br

sakaki-, 2019-08-29:
Yes, it is possible as the image now supports both RPi4 and RPi3 (just a little fiddly, as you will need to expand the size of the `/boot` partition prior to upgrading, using a tool like `gparted`, since there are now two kernels to accommodate).

Please see my manual upgrade instructions [here](https://github.com/sakaki-/gentoo-on-rpi-64bit/releases/#genup_to_1_5_0).

sakaki-, 2019-09-06:
Closing now as I believe this point is covered.
Best, sakaki

closed 2019-09-06 by sakaki-