sakaki-/gentoo-on-rpi-64bit #104: ARM errata workarounds via the alternatives framework

Avamander, 2019-09-07:
After following the "Build your own kernel" guide and entering menuconfig, I noticed that a lot of workarounds under `ARM errata workarounds via the alternatives framework` are enabled, even though the chipset is only Cortex-A53, is this intentional? If not, are the workarounds applied on the binary kernel? If yes, maybe they could be disabled?

jfikar, 2019-09-08:
For the second question I think Gentoo looks into /usr/src/linux. This should be a symlink to your kernel directory.

Avamander, 2019-09-09:
The kernel I compiled where I disabled all the workarounds except the Cortex-A53 ones works nicely, will continue using it to see how stable it is, but I do think they're wrongly enabled by default bcm config.

sakaki-, 2019-09-09:
That's correct about `/usr/src/linux`.

The binary kernel's configuration is based on `bcmrpi3_defconfig` on the RPi3 B/B+ and `bcm2711_defconfig` for the RPi4 B (autobuilds [here](https://github.com/sakaki-/bcmrpi3-kernel-bis) and [here](https://github.com/sakaki-/bcm2711-kernel-bis)). Those configs come from [upsteam](https://github.com/raspberrypi/linux/tree/rpi-4.19.y/arch/arm64/configs).

As such, why not post an issue about this to [raspberrypi/linux](https://github.com/raspberrypi/linux) to see their response on this point.?

Best, sakaki

necrose99, 2020-02-27:
I have rockpro64 , and rpi3 , looking to get rpi4 also and pinebook pro... 
anyrate , rp64 i got and 4 month latter after i got it pi4 droped.. 

however for common want of kernel, i make all modules , make all yes. , 
slam aufs5 built iin via standalone and zfs > , 

I prune sys-fs from config. , sys-fs-tools ok , but next general setup dep...sys-fs-v2 i sed out or make menu config.  gentoo-sources pentoo-sources , with panfrost/mali dashed in zfs /aufs5 
ie make mennu after can pick opts on , while having 99% of what i'd do w/o 3-4 hours y n module uggh.  having to comb for a few patches , as bluetooth drivers are simular but not on same ic2 etc bus .  

(config.bad /config.good cat my.config make oldconfig , or another way of taking shards merging. 
ie conf.1 2. 3. etc some distros merger many to one .config.  ) -- while very program-tic i feal a need of aspirin.. already. 

mono kernel, not the greatest fan but **one kernel too**( boot arm64 server/pi sbc's etc..) **Rule them all.**  other than uboot per device.  , efi , grub i have precompiled zfs uboot efi yep. 

VC4+Panfrost+MaLI= same fram buffer x-11 , fbtubrb , just mesa needs to pplay nicer. 
if i had wild hairs nuevo for an older geforce but i'd rather use m.2 
and pool packages distfiles etc on rockpro64 

I have grub built , 

**next relaese of RPI4/3** 

/boot/efi   , leaders overlay etc...   rpi4-boot.me , rock64.bootme etc.. 
./efi/gentoo/grub-aarch64.efi
if i need to swap to other system cool. renam foo.bak , rename. to which board going to run. 
rock64./pro can burn uboot to onboard rom after getting Linux  up. 
mirnor irritation of using manjaro image , then vacating it. (_keeping loader/s/uboot parts_)  , <<< gentoo stage4 tarball , fix boot. etc.  chroot add/remove pkgs , recompile wala. 

/boot/ , kernel/s like Debian 

getting rpi4/3 rock64/pro /pinebook device trrees in raw and atf is a minor pain...

more recent uboot suposedly can work on gpt cards , a nicety if rpi did for sure.

closed 2022-11-28 by Avamander