sakaki-/gentoo-on-rpi-64bit #105: Raspberry Pi metapackage depends on openrc/sysvinit

Avamander, 2019-09-10:
When I was converting my system to use `systemd` I ran into the problem that the `rpi-64bit-meta` package depends on `openrc`/`sysvinit` which obviously causes conflicts with `systemd`. It would be nice if the metapackage could detect the systemd use flag and act accordingly.

Thanks again for the base image and awesome readme!

sakaki-, 2019-09-10:
Thanks, good catch.
I'll look to fix that next time I bump the metapackage.
Best, sakaki

Avamander, 2019-09-14:
There's also this package:
```
# required by net-wireless/rpi3-wifi-regdom-1.1::genpi64
# required by rpi3-wifi-regdom (argument)
# /etc/portage/package.mask/mask:
=sys-apps/openrc-0.42.1-r2
# required by sys-apps/openrc-0.42.1-r2::genpi64
# required by net-wireless/rpi3-wifi-regdom-1.1::genpi64
# required by rpi3-wifi-regdom (argument)
# /etc/portage/package.mask/mask:
=sys-apps/sysvinit-2.95-r1
```

Avamander, 2019-09-14:
I ran genup and it started removing a lot of packages and it seems that a lot more packages don't want to work with systemd. I'm listing them so that you don't have to search that much if you decide to take a look.

For example these two:
```
# required by sys-apps/rpi3-init-scripts-1.1.5::rpi3
# required by rpi3-init-scripts (argument)
# /etc/portage/package.mask/mask:
=sys-apps/openrc-0.42.1-r2
```

```
# required by sys-apps/openrc-0.42.1-r2::genpi64
# required by net-misc/netifrc-0.6.1::gentoo
# /etc/portage/package.mask/mask:
=sys-apps/sysvinit-2.95-r1
```

sakaki-, 2019-09-14:
Thx!

I have nothing against `systemd` as such btw (see for example my recent [raspbian-nspawn-64](https://github.com/sakaki-/raspbian-nspawn-64) image update), so if you create `systemd` units for the various rpi-specific services on the lite image, feel free to PM me, and I'll be happy to review/include them. It's just that I don't have the bandwidth to support two separate init system images atm.

Best, sakaki

Avamander, 2019-09-19:
Well, I'd just love to see most of the packages not pulling in sysvinit/openrc if `systemd` USE flag is specified. Though the `cmdline.txt` file could use a systemd version by just adding 

Also, reinstalling Pulseaudio, it complained that:
```
 * Checking for suitable kernel configuration options...                                                                                                   [ ok ]
 * A preallocated buffer-size of 2048 (kB) or higher is recommended for the HD-audio driver!
 * CONFIG_SND_HDA_PREALLOC_SIZE=64
```
Given my kernel is basically default Pi3 config with only a few errata disabled, it would make sense to rise this for everyone.

sakaki-, 2019-11-24:
Hopefully fixed via https://github.com/sakaki-/genpi64-overlay/commit/6b3f2e076a9482cf8e013cddcbb5a82734a4918d
PRs to add the necessary systemd units welcome ^-^

Best, sakaki

PS `CONFIG_SND_HDA_PREALLOC_SIZE` has also been set by default for the `-bis` kernels (see https://github.com/sakaki-/bcm2711-kernel-bis/commit/dd093579f7666a5c229b2913d00f383734ea969e and https://github.com/sakaki-/bcmrpi3-kernel-bis/commit/37c886e25832a4fefbb9a973dcc2c0e2345d5d06).

sakaki-, 2019-11-24:
Closing now

closed 2019-11-24 by sakaki-