sakaki-/gentoo-on-rpi-64bit #107: CONFIG_KEY_DH_OPERATIONS required

skulko, 2019-09-16:
Hello

sys-apps/keyutils-1.6.1 may need CONFIG_KEY_DH_OPERATIONS=y

maybe you can change this for the kernels here?!

best regards

sakaki-, 2019-09-16:
Yes, please just file a PR against [`bcm2711-kernel-bis`](https://github.com/sakaki-/bcm2711-kernel-bis) (for the RPi4 kernel) and [`bcmrpi3-kernel-bis`](https://github.com/sakaki-/bcmrpi3-kernel-bis) (for the RPi3 kernel).
Example PR may be seen [here](https://github.com/sakaki-/bcm2711-kernel-bis/pull/1).
Thanks, sakaki

sakaki-, 2019-09-17:
Thanks for the PRs. These seem to have taken OK in the latest autobuilds (see e.g. [here](https://github.com/sakaki-/bcm2711-kernel-bis/blob/37153f57507963f73b2b2c94bccfd02817ad228c/vs_bcm2711_config.diff#L538-L540) and [here](https://github.com/sakaki-/bcmrpi3-kernel-bis/blob/a13b428277225ebd3998e96bbc403f7d84ed9c67/vs_bcmrpi3_config.diff#L121-L123)).

So, to use them, just:

```console
demouser@pi64 ~ $ emaint sync --repo genpi64
demouser@pi64 ~ $ echo "~sys-kernel/bcm2711-kernel-bis-bin-4.19.73.20190917" | sudo tee -a /etc/portage/package.unmask/bcm2711-kernel-bis-bin
demouser@pi64 ~ $ echo "~sys-kernel/bcmrpi3-kernel-bis-bin-4.19.73.20190917" | sudo tee -a /etc/portage/package.unmask/bcmrpi3-kernel-bis-bin
demouser@pi64 ~ $ sudo emerge -v1u bcm2711-kernel-bis-bin bcmrpi3-kernel-bis-bin
```

Reboot once complete, and you should hopefully be using your new kernel (with `CONFIG_KEY_DH_OPERATIONS` enabled).

Best, sakaki

skulko, 2019-09-18:
Awesome!
Thanks a lot everything.

closed 2019-09-18 by skulko