sakaki-/gentoo-on-rpi-64bit #108: code linked to gpiomem (including wiringPi) need root access to run

arizonadrscott, 2019-09-16:
Your Gentoo arch64 system is working very well and seems very stable.  However, any code related to gpio can not except for root access.  To solve this, I recommend the following:

Create NEW device groups...

sudo groupadd spi
sudo groupadd i2c
sudo groupadd gpio

Add groups to demouser account:
sudo usermod -a -G i2c demouser
sudo usermod -a -G spi demouser
sudo usermod -a -G spi demouser

add these udev rules to /etc/udev/rules.d directory:
( create a 99-com.rules file as in current rasbian image)

SUBSYSTEM=="input", GROUP="input", MODE="0660"
SUBSYSTEM=="i2c-dev", GROUP="i2c", MODE="0660"
SUBSYSTEM=="spidev", GROUP="spi", MODE="0660"
SUBSYSTEM=="bcm2835-gpiomem", GROUP="gpio", MODE="0660"

SUBSYSTEM=="gpio", GROUP="gpio", MODE="0660"
SUBSYSTEM=="gpio*", PROGRAM="/bin/sh -c '\
	chown -R root:gpio /sys/class/gpio && chmod -R 770 /sys/class/gpio;\
	chown -R root:gpio /sys/devices/virtual/gpio && chmod -R 770 /sys/devices/virtual/gpio;\
	chown -R root:gpio /sys$devpath && chmod -R 770 /sys$devpath\
'"

KERNEL=="ttyAMA[01]", PROGRAM="/bin/sh -c '\
	ALIASES=/proc/device-tree/aliases; \
	if cmp -s $ALIASES/uart0 $ALIASES/serial0; then \
		echo 0;\
	elif cmp -s $ALIASES/uart0 $ALIASES/serial1; then \
		echo 1; \
	else \
		exit 1; \
	fi\
'", SYMLINK+="serial%c"

KERNEL=="ttyS0", PROGRAM="/bin/sh -c '\
	ALIASES=/proc/device-tree/aliases; \
	if cmp -s $ALIASES/uart1 $ALIASES/serial0; then \
		echo 0; \
	elif cmp -s $ALIASES/uart1 $ALIASES/serial1; then \
		echo 1; \
	else \
		exit 1; \
	fi \
'", SYMLINK+="serial%c"

Verify that everything works as follows:

demouser@pi64 ~ $ groups
adm disk lp wheel cron audio cdrom video usb users plugdev portage demouser gpio spi i2c
demouser@pi64 ~ $ ls -al /dev/spi*
crw-rw---- 1 root spi 153, 0 Sep 16 10:21 /dev/spidev0.0
crw-rw---- 1 root spi 153, 1 Sep 16 10:21 /dev/spidev0.1
demouser@pi64 ~ $ ls -al /dev/i2c*
crw-rw---- 1 root i2c 89, 1 Sep 16 10:21 /dev/i2c-1
demouser@pi64 ~ $ ls -al /dev/gpio*
crw-rw---- 1 root gpio 254, 0 Dec 31  1969 /dev/gpiochip0
crw-rw---- 1 root gpio 254, 1 Sep 16 10:21 /dev/gpiochip1
crw-rw---- 1 root gpio 254, 2 Dec 31  1969 /dev/gpiochip2
crw-rw---- 1 root gpio 245, 0 Dec 31  1969 /dev/gpiomem

sakaki-, 2019-09-29:
Hi @arizonadrscott,

so, currently the `rpi3-i2cdev` and `rpi3-spidev` packages do set up `udev` rules (in `/lib/udev/rules.d/`) to allow `wheel` group access for their respective interfaces, but the approach you suggest is probably better (and is more general).

I'll leave this issue open as a reminder to address this on the next point release.

Thanks for the suggestion,

best, sakaki

sakaki-, 2019-11-27:
Implemented (see [`sys-apps/rpi-gpio`](https://github.com/sakaki-/genpi64-overlay/tree/master/sys-apps/rpi-gpio), [`sys-apps/rpi-i2c`](https://github.com/sakaki-/genpi64-overlay/tree/master/sys-apps/rpi-i2c) and [`sys-apps/rpi-spi`](https://github.com/sakaki-/genpi64-overlay/tree/master/sys-apps/rpi-spi); also [`sys-apps/rpi-serial`](https://github.com/sakaki-/genpi64-overlay/tree/master/sys-apps/rpi-serial) and [`sys-apps/rpi-video`](https://github.com/sakaki-/genpi64-overlay/tree/master/sys-apps/rpi-video)).

Thanks for the suggestion!

closed 2019-11-27 by sakaki-