sakaki-/gentoo-on-rpi-64bit #109: no wifi on latest release

Askmewho, 2019-09-20:
just that, ive set the wifi location, everything, but ive only get wifi working with another usb wifi adapter

offtopic: your work its remarkable, gentoo its horrible.

sakaki-, 2019-09-20:
Strange that WiFi isn't working for you; there were locale issues with 1.5.0, but I had thought these were fixed on 1.5.1, and the users who had problems (e.g. Gazzy, roylongbottom, jfikar etc.) report that the new release works fine for WiFi.

Can you see your access point in the NetworkManager menu? How strong is the signal?

Does it work with vanilla (32-bit kernel, 32-bit userland) Raspbian on the same board, in the same location?
If so, does it work with my [raspbian-nspawn-64](https://github.com/sakaki-/raspbian-nspawn-64) image (as that has a 64-bit kernel but 32-bit userland)?

Best, sakaki

Askmewho, 2019-09-21:
my mistake, sorry! mine its not the latest!

closed 2019-09-21 by Askmewho