sakaki-/gentoo-on-rpi-64bit #111: why DEFAULT portage tree is from "together".org?

ickromwerk, 2019-09-24:
according to /etc/portage/repos.conf/gentoo.conf:
```
[DEFAULT]
main-repo = gentoo

[gentoo]
location = /usr/portage
sync-type = rsync
#sync-uri = rsync://rsync.gentoo.org/gentoo-portage
sync-uri = rsync://isshoni.org/gentoo-portage-pi64-gem
auto-sync = yes
```

isshoni.org http says:
```
Welcome to isshoni.org!
This site contains a number of my portage binhosts.
Use the packages here at your own risk!
```

doesn't genpi64 overlay is not enough?

sakaki-, 2019-09-24:
Hi @ickromwerk,

that's because, as discussed [here](https://github.com/sakaki-/gentoo-on-rpi-64bit#binhost), I host a _time-gated_ version of the Portage tree from isshoni.org, to ensure that when users update, to the maximum extent possible there are binary packages already available for any bumped versions; that that necessary masks, USE flags etc. have already been set in the overlay etc.

Since the gated tree still uses Portage's `gemato` signature verification (and this is checked on sync), you can be assured that it has not been modified from upstream (other than the delay).

If you want, you can always switch to the 'live' tree, but in this case you'll need to be prepared for some (very) large local (from-source) emerges, for example, when `chromium` or `libreoffice` bumps.

hth, sakaki

ickromwerk, 2019-09-24:
That means that if I uncomment official repo and comment isshou one it should work fine without issues?

Is it possible to use official tree and intentionally mask large package to certain versions in portage.mask? Then genup will bump masks file to binary compiled versions. Then if user don't want to wait for bindist he can unmask certain "biggie" and compile by yourself.

ickromwerk, 2019-09-24:
Like /etc/portage/package.mask/binary_compiled_pi64:

app-office/libreoffice >17.3.6
...

And that versions are that have binary compiled packages on isshou.org
And add them to bindist location

ickromwerk, 2019-09-24:
In global view it might look like this:

1. When you update bindist on your site you generate mask record for each compiled package.
2. When user runs genup it gets latest official portage tree and rpi64 overlay. Then it sync portage.mask of compiled versions.

That way user always has latest tree even if time gap between it and compiled versions is high for some reasons and if he don't want to wait he can manually unmask certain package and compile it from sources.

sakaki-, 2019-09-29:
Thanks for the suggestion!

At this stage, I probably won't transition to this approach however, since just masking back isn't generally sufficient. That's because old ebuilds are often _dropped_ from the main tree when revision or version bumped, so to avoid the situation where _no_ available ebuild is valid (without local unmasking) for a given package, the 'last known good' ebuild would also need to be copied into e.g. the `genpi64` overlay, which makes for quite a large delta each time.

I've tended to find that most users are either happy with the lagged-tree version, or immediately transition to the stock upstream tree for full control. There's been a bit of a delay updating the lagged copy tree since the v1.5.1 release, since I've been doing some work on the build server infrastructure (apologies), but an update will hopefully be pushed out within a day or so, and I try to keep the update cadence biweekly (at worst) in normal running (and therefore not too far in real terms from the live tree).

I'll close this issue now; hope that's OK.

Best, sakaki

closed 2019-09-29 by sakaki-