sakaki-/gentoo-on-rpi-64bit #112: TeamViewer on Gentoo

callmehomer, 2019-09-30:
Hello
I'm trying to install TeamViewer via Emerge. However I'm getting some packages error.

`demouser@pi64 ~ $ emerge --ask net-misc/teamviewer

Local copy of remote index is up-to-date and will be used.
This action requires superuser access...
Would you like to add --pretend to options? [Yes/No] yes
 * Last emerge --sync was 37d 19h 38m 59s ago.

These are the packages that would be merged, in order:

Calculating dependencies... done!
[binary  N     ] dev-qt/qtdeclarative-5.12.4  USE="widgets -debug (-gles2) (-jit) -localstorage -test" 
[binary  N     ] dev-qt/qtquickcontrols-5.12.4  USE="widgets -debug -test" 
[binary  N     ] dev-qt/qtwebkit-5.212.0_pre20180120  USE="X gstreamer hyphen opengl printsupport -geolocation (-gles2) (-jit) -multimedia -nsplugin -orientation -qml -webp" 
[ebuild  N    *] net-misc/teamviewer-14.4.2669 
**[blocks B      ] sys-auth/consolekit ("sys-auth/consolekit" is blocking net-misc/teamviewer-14.4.2669)**

 * Error: The above package list contains packages which cannot be
 * installed at the same time on the same system.

  (sys-auth/consolekit-1.2.1:0/0::gentoo, installed) pulled in by
    sys-auth/consolekit[pam] required by (sys-auth/pambase-20190402:0/0::gentoo, installed)
    sys-auth/consolekit required by (x11-apps/xdm-1.1.12:0/0::gentoo, installed)
    >=sys-auth/consolekit-1.0.0 required by (net-misc/networkmanager-1.18.2:0/0::gentoo, installed)
    sys-auth/consolekit[policykit] required by (sys-auth/polkit-0.115-r3:0/0::gentoo, installed)

  (net-misc/teamviewer-14.4.2669:14/14::gentoo, ebuild scheduled for merge) pulled in by
    net-misc/teamviewer

For more information about Blocked Packages, please refer to the following
section of the Gentoo Linux x86 Handbook (architecture is irrelevant):

https://wiki.gentoo.org/wiki/Handbook:X86/Working/Portage#Blocked_packages

demouser@pi64 ~ $`

I'm not Linux pro user and I can't figure out how to solve it. Any tips?

Links:
https://packages.gentoo.org/packages/net-misc/teamviewer
https://community.teamviewer.com/t5/Knowledge-Base/How-to-install-TeamViewer-on-other-Linux-distributions/ta-p/30660
https://pastebin.com/SFkeawri

sakaki-, 2019-09-30:
The above output is telling you that the `teamviewer` ebuild refuses to be installed on a system (like yours) which has `consolekit` present (and which, therefore, 'blocks' it). The relevant line is [this one](https://github.com/gentoo/gentoo/blob/2cbc944548b2e15a3a39561968b4c9acbb8469b5/net-misc/teamviewer/teamviewer-14.4.2669.ebuild#L31).

The _underlying_ reason the block was put there by the ebuild author, is because the package (presumably) wants either full `systemd`, or `elogind` (systemd's logind extracted to a standalone package) to manage login sessions etc., rather than `consolekit`. See [later in the ebuild](https://github.com/gentoo/gentoo/blob/2cbc944548b2e15a3a39561968b4c9acbb8469b5/net-misc/teamviewer/teamviewer-14.4.2669.ebuild#L129-L131) for some further colour on this.

This is really an upstream issue: if you post a question to the [Gentoo forums](https://forums.gentoo.org/viewforum-f-62.html) about installing `teamviewer` on a system with `OpenRC` init and `consolekit` installed, someone may be able to assist.

> Note that in general, switching over to `elogind` is somewhat easier than a wholesale switch to `systemd`. Any of these alternatives is possible within Gentoo, but will require some recompilation work (as they aren't the native setup for this image).

hth, sakaki

ykla, 2019-11-20:
net-misc/teamviewer seems not support arm

sakaki-, 2019-11-21:
As mentioned, the best course here would be to post in the Gentoo forums, since this is really an upstream issue.
Closing this now.

closed 2019-11-21 by sakaki-