sakaki-/gentoo-on-rpi-64bit #113: Difficulty updating locale to ja_JP

ajping, 2019-10-07:
I've followed the instructions and the locale changes to Japan/Japanese. But for some reason when I reboot the change is not permanent and it reverts back to GB locale. Apologies if this is a noob question - I don't have much experience with localization.

closed 2019-10-07 by ajping

sakaki-, 2019-10-07:
Hi - did you get this solved?
Best, sakaki

ajping, 2019-10-07:
Yes, I think so. It turns out that setting locale with env-update only
writes LANG in 02locale. But that wasn't enough to keep the system switched
over to Japanese - something kept switching it back to GB. I manually added
and set LC_CTYPE to ja_JP.utf8 and that keeps it switched over. After
adding and setting LC_TIME, the date command now shows the correct
localtime as well!

Best, Aaron

sakaki-, 2019-10-07:
Glad to hear you got it sorted ^-^
I'll try and add a wiki page about localization shortly, had quite a few queries about it recently.
Best, sakaki