sakaki-/gentoo-on-rpi-64bit #115: portage runs very slow so that it cannot install any softwares

ykla, 2019-10-18:
For raspberry pi 4b +, OS v 15.
Portage runs very slow. I can't install any softwares. Just likes emerge --ask screen.

sakaki-, 2019-10-18:
Make sure you have a working network connection to your Pi4, as the first thing `emerge` does is to synchronize the binary package catalogue (and will hang waiting for this - until quite a long timeout is reached - if the system is offline).

Also, Portage has to calculate dependencies for any install, and this can take a while (given the multiple versions, USE flags etc. that Gentoo allows). It is always going to be slower than pure-binary package manager like `apt` etc.

Try something simple to ensure everything is basically working e.g.:

```console
demouser@pi64 ~ $ sudo emerge -v dev-util/diffstat
```

That should complete within a few minutes. If not, something is definitely wrong.

To speed up the actual _building_ of packages, you can open `/etc/portage/make.conf` and uncomment the lines specifying `MAKEOPTS` and `EMERGE_DEFAULT_OPTS` (and save).

ykla, 2019-10-20:
In fact， I have uncommented them in make.conf, but no useful.

`pi64 ~ # time emerge -v dev-util/diffstat
^C

Exiting on signal Signals.SIGINT

real    367m20.778s
user    0m3.999s
sys     0m0.204s
pi64 ~ # 
`

No response at all.
And I changed the download sources so that I and cross GFW,
`pi64 ~ # ping bing.com
PING bing.com (204.79.197.200) 56(84) bytes of data.
64 bytes from a-0001.a-msedge.net (204.79.197.200): icmp_seq=1 ttl=111 time=78.2 ms
64 bytes from a-0001.a-msedge.net (204.79.197.200): icmp_seq=2 ttl=111 time=75.4 ms
64 bytes from a-0001.a-msedge.net (204.79.197.200): icmp_seq=3 ttl=111 time=76.2 ms
^C
--- bing.com ping statistics ---
3 packets transmitted, 3 received, 0% packet loss, time 2002ms
rtt min/avg/max/mdev = 75.393/76.604/78.175/1.163 ms
pi64 ~ # 
`
There is my make.conf:

`pi64 ~ # cat /etc/portage/make.conf 
# Simple make.conf for 64-bit Raspberry Pi 3 & Pi 4 SBCs

# NB most settings now taken from the default/linux/arm64/17.0/desktop/genpi64
# profile
# See /usr/local/portage/genpi64/profiles/targets/genpi64/<...>
# You can override as desired in this file (/etc/portage/make.conf)
# (and via the other /etc/portage/<...> subdirectory entries)

# global USE flag overrides
USE=""

# some scripts looks for PORTDIR in make.conf explicitly
PORTDIR="/usr/portage"

# override default build parallelism by uncommenting the following
# two lines, and modifying the values as required
# Pi4 users (particularly those with a 4GiB variant) can afford to be
# more aggressive here
MAKEOPTS="-j5 -l4"
EMERGE_DEFAULT_OPTS="--jobs=5 --load-average=4"
# caution! many parallel threads and / or emerge jobs may cause your
# RPi3/4 to run out of memory during large builds, or genup runs

# per https://wiki.gentoo.org/wiki/Raspberry_Pi_VC4
VIDEO_CARDS="fbdev vc4"
INPUT_DEVICES="evdev synaptics"

# uncomment to build binary packages as a byproduct of each emerge
# (these are useful backups) in /usr/portage/packages
FEATURES="${FEATURES} buildpkg"

# uncomment to disribute emerges, where possible, using distcc
#FEATURES="${FEATURES} distcc distcc-pump"

# uncomment to use binary packages from PORTAGE_BINHOST, where available,
# (and build normally, where not)
FEATURES="${FEATURES} getbinpkg"
PORTAGE_BINHOST="https://isshoni.org/pi64pie"

GENTOO_MIRRORS="https://mirrors.ustc.edu.cn/gentoo"
pi64 ~ # 
`

And OS is new install.

sakaki-, 2019-10-20:
`isshoni.org` is on a fixed IP, `82.221.139.201` (as you have already determined), but the server had inbound pings dropped by default. I've enabled them now, so you should be able to `ping` if you try.

Also, if you run `wget isshoni.org`, and this can successfully download an `index.html` file from this site, everything else should work too, so please try that.

If you _can_ reach the site, there's not much point in mirroring it locally, but if you want to, this is of course possible using tools like `wget` per various tutorials available online.

Please let me know if this is really what you need to do, and I can point you in the right direction.

Best, sakaki

ykla, 2019-10-21:
Sorry I don't know why once I uncomment line
PORTAGE_BINHOST="https://isshoni.org/pi64pie" 
that I can run emerge, or It not run at all.

sakaki-, 2019-10-21:
Hi @ykla,

The reason that uncommenting the line makes it work, is that `getbinpkg` is set by default in `FEATURES` (via `/etc/portage/make.conf`, [here](https://github.com/sakaki-/gentoo-on-rpi-64bit/blob/6c9a7582f47128cc020d1ee2e043eb13f1b215aa/reference/make.conf#L37)). This means `emerge` is always going to try to synchronize with the binary package host first. Comment out _both_ lines, or leave both of them uncommented, in `make.conf`; so _either_:

```bash
FEATURES="${FEATURES} getbinpkg"
PORTAGE_BINHOST="https://isshoni.org/pi64pie"
```

to use the binhost where it has precompiled packages available that match your USE flag settings etc, _or_

```bash
#FEATURES="${FEATURES} getbinpkg"
#PORTAGE_BINHOST="https://isshoni.org/pi64pie"
```
to always build locally, and ignore the binhost completely.

hth, sakaki

sakaki-, 2019-11-10:
Closing now - please re-open if still experiencing issues with this.
Best, sakaki

closed 2019-11-10 by sakaki-