sakaki-/gentoo-on-rpi-64bit #116: upnp with kodi

fsamson91, 2019-10-18:
I have installed this gentoo img on my rpi, and it 's a great linux server in 64 bits. I never used gentoo,  I 'm using ubuntu on my server. So I want to use kodi on my rpi with gentoo and kodi seems to be configured without upnp protocole. How can I had this protocol ? 

Thanks

sakaki-, 2019-10-18:
Hi @fsamson91,

to use UPnP with Kodi, you'll need to recompile it with the relevant USE flag set (here, `upnp`, which is off by default).

To do so, issue (as root, or use `sudo`):
```console
pi64 ~ # echo "media-tv/kodi upnp" >> /etc/portage/package.use/kodi
```

> You can read more setting per-package USE flags in Gentoo etc. in my notes [here](https://wiki.gentoo.org/wiki/Sakaki%27s_EFI_Install_Guide/Installing_the_Gentoo_Stage_3_Files#Portage.27s_Configuration_Files).

Once that's done, for efficiency edit the file `/etc/portage/make.conf` and _uncomment_ the `MAKEOPTS` and `EMERGE_DEFAULT_OPTS` lines (to enable optimal build parallelism). Save, then re-emerge `kodi`, to build it with your new custom USE flag set:

```console
pi64 ~ # emerge -av media-tv/kodi
```

This may take a while - it's a largish package ><

If you do build it and it works OK, please let me know, and I'll consider adding `upnp` to the (additional-to-default) [profile USE flags](https://github.com/sakaki-/genpi64-overlay/blob/master/profiles/targets/rpi3/package.use/kodi) for this package on the binhost.

hth, sakaki

fsamson91, 2019-10-21:
Hi, sakaki

I tried your commands and after few hours, I run kodi but the upnp is not available I try again the emerge to have the output, and here are the message :

[ebuild     U *] media-tv/kodi-18.4::gentoo [18.0::gentoo] USE="X alsa bluetooth css dbus lcms opengl pulseaudio samba (system-ffmpeg) udev udisks upnp* upower xslt -airplay -bluray -caps -cec (-dvd) -gbm -gles -libressl -libusb (-lirc) -mariadb -mysql -nfs (-raspberry-pi) -systemd -test -vaapi (-vdpau) -wayland -webserver -zeroconf (-debug%)" PYTHON_TARGETS="python2_7" 0 KiB

Total: 1 package (1 upgrade), Size of downloads: 0 KiB

!!! The following binary packages have been ignored due to non matching USE:

    =media-tv/kodi-18.4 -upnp

NOTE: The --binpkg-respect-use=n option will prevent emerge
      from ignoring these binary packages if possible.
      Using --binpkg-respect-use=y will silence this warning.

So I don't know wich package is not matching..

Franck

sakaki-, 2019-10-21:
Hi @fsamson91,

so, the above output shows that you are currently on version 18.0 of Kodi (`[18.0::gentoo]`), and that the `emerge` wants to **U**pgrade to 18.4 by building locally (`[ebuild U ] media-tv/kodi-18.4::gentoo`), and this local build is happening because it can't use the precompiled version on the binhost, since the USE flags don't match:

```
!!! The following binary packages have been ignored due to non matching USE:

    =media-tv/kodi-18.4 -upnp
```

(which is to be expected, since you have added the `upnp` USE flag locally).

However, this also implies that your previous `emerge` did _not_ complete successfully (since otherwise you'd already be at 18.4, and the second `emerge` would have signalled a **R**einstall (not **U**pgrade).

Take a look at the end of `/var/log/emerge.log` to see what happened to the previous `emerge` attempt. As I say, building `kodi` locally from source (without e.g. [`distcc` acceleration](https://github.com/sakaki-/gentoo-on-rpi-64bit/wiki/Set-Up-Your-crossdev-PC-for-Distributed-Compilation-with-distcc)), may take many hours to complete. You can monitor ongoing builds by using the `showem` command in another terminal window, which an be helpful.

hth, sakaki

fsamson91, 2019-10-23:
Hi Sakaki, 

I tried again with a reinstall from scratch and the emerge command and after many many hours gentoo said that the kernel sources are need because the it multicast is need for kodi. 
I will try to install kernel sources later... 

Franck

sakaki-, 2019-10-23:
Hi Franck,

so, I have [just added](https://github.com/sakaki-/genpi64-overlay/commit/066bfc131c3301a7855a3760c2813efe0db4bdf4) the `upnp` flag on the custom profile and rebuilt `kodi` on the binhost. So, to get it (from binary package), just run:

```console
pi64 ~ # emaint sync --repo genpi64
pi64 ~ # emerge -v1u media-tv/kodi
```

It should hopefully install from binary now.

> If you have any issues, just run `genup` to fully update your system; this should be left running overnight, as it will take a while, but hopefully, you won't need to use it.

Pls let me know if that works for you.

PS you can safely delete your `/etc/portage/package.use/kodi` file, as the profile now sets this flag.