sakaki-/gentoo-on-rpi-64bit #118: rpi config tool for the command line

colllin3639, 2019-11-01:
Would it be possible to have the rpi config tool for the command line

sakaki-, 2020-01-31:
It would be possible, but it is really just a way of manipulating `/boot/config.txt` for the most part, and people comfortable with using the CLI (and/or the 'lite' version of the image) are also (in my experience) generally happier (and certainly capable) of editing this by hand.

I'll mark this as an enhancement / help wanted for now, as someone may be willing to jump in here and it'd be a nice orthogonal project to add into the image.

JJTech0130, 2020-03-01:
Just saw this, it would be nice if we had an "easy" way to setup the wireless network from the command line. Would this be possible? (Or did you add a networking tutorial I missed?) Similar to what you can do in the regular rpi-config.

sakaki-, 2020-03-01:
Hi @JJTech0130 

`nmtui` is bundled (even with the lite image) and this tool makes it relatively straightforward to set up WiFi networking from the command line.

Happy to add a tutorial for this tool if you think it'd be useful?

best, sakaki

JJTech0130, 2020-03-02:
Thanks, I didn't know about this tool! (I just got started with gento :smile: )
It would probably be helpful if you make a mini-tutorial for the CLI as I am trying to spped up my server by using 64bit, and I don't know much about gentoo. (I cant use the gui cuz its a server & the network setup in the GUI only starts with the GUI anyways)