sakaki-/gentoo-on-rpi-64bit #121: port open

mcsim415, 2019-12-01:
How to open port 19132?

sakaki-, 2019-12-01:
Hello @mcsim415,

sorry, I don't quite understand what you are asking here? The image, as shipped, does not actively firewall inbound connections (although this can of course be turned on easily enough).

Is there a particular service, using this port, that you'd like to enable?

Best, sakaki

mcsim415, 2019-12-01:
I couldn't find service command.
How can i do?
[image]

sakaki-, 2019-12-02:
Try `rc-service iptables save`?

mcsim415, 2019-12-02:
Oh. It's working.
[image]

mcsim415, 2019-12-02:
Um.. It isn't working..
[image]
[image]

mcsim415, 2019-12-02:
22 port is open.
But 19132 port is close..

mcsim415, 2019-12-02:
In fact, I can't use Linux well..

sakaki-, 2019-12-02:
You need a service listening on this port also. Try running `netstat -vatn` and `netstat -vaun` to see which TCP and UDP services are running, on which ports.

mcsim415, 2019-12-02:
[image]

mcsim415, 2019-12-02:
I want open my minecraft pe server!

mcsim415, 2019-12-02:
Ummm

sakaki-, 2019-12-02:
OK, from your `netstat` output, you have a UDP service listening on port 19132, presumably [pocket minecraft protocol](https://wiki.vg/Pocket_Minecraft_Protocol).

As such, you need to allow _inbound_ connections on that port for UDP (rather than outbound connections for TCP, which is what you did above).

Try:

```console
# iptables -P OUTPUT ACCEPT
# iptables -I INPUT 1 -p udp --dport 19132 -j ACCEPT
```
verify your ruleset with
```console
# iptables-save
```

hth, sakaki

mcsim415, 2019-12-03:
Ummm.?
[image]

sakaki-, 2019-12-03:
OK, so as I suspected, the port _is_ open (you have in fact many rules ^-^ inserted to do that, and also, the *default* stance for `INPUT` is `ACCEPT` too; so nothing on this local firewall is in fact blocking you &mdash; which is as I would have expected, unless you had explicitly set up something to the contrary).

You also (per your `netstat -vaun` output [earlier](https://github.com/sakaki-/gentoo-on-rpi-64bit/issues/121#issuecomment-560326539)) have a service listening on this port, on all interfaces.

So you should be able to connect to this service OK. Check the IP address of your RPi using `ifconfig`, and then try to connect from another system on your local network first. Once that works, you can ensure port forwarding etc. on your external router is working.

mcsim415, 2019-12-03:
no... Port isn't open..
[Go here and leamo.kro.kr;19132](https://www.yougetsignal.com/tools/open-ports/)

mcsim415, 2019-12-03:
I'll check it port state soon.

mcsim415, 2019-12-03:
Haaa
[image]

mcsim415, 2019-12-03:
Oh my gosh...

mcsim415, 2019-12-03:
It is still close

sakaki-, 2019-12-03:
The issue is likely with your external router allowing / forwarding port 22, but not your target port. You need to fix that. The issue is not with the Pi.

Anyway, I am closing this issue and blocking you as a user, since [your post above](https://github.com/sakaki-/gentoo-on-rpi-64bit/issues/121#issuecomment-561089008) contains inappropriate sexual imagery.

I have reported this also to GitHub.

S.

closed 2019-12-03 by sakaki-