sakaki-/gentoo-on-rpi-64bit #127: net-dialup/freeradius-3.0.17 missing /usr/lib64/rlm_mschap.so

seanuxbridgeperluser, 2019-12-15:
Hello,

FreeRADIUS will not run.  When I try to use usr/sbin/radiusd, it complains that /usr/lib64/rlm_mschap.so is missing.

I have three different suggestions on how to resolve.

1.) I don't use MS-CHAP, so if you need to build without linking to it, that's fine.
2.) You can release a version of 3.0.17 that has the required library
3.) You can release a newer version as stable that has the required library
4.) Or, heck, there could be another idea that I'm not thinking of

I just really want to run RADIUS on my Pi.  Can you help?

(also, that's a dynamically-linked binary under sbin but that's neither here nor there)

sakaki-, 2019-12-16:
Looking at the ebuild for this package, it seems you need to set the `samba` USE flag [to retain](https://github.com/gentoo/gentoo/blob/0f8f02e2d863c8a591ea9bdb1a623527ec2313ba/net-dialup/freeradius/freeradius-3.0.17.ebuild#L89-L90) this library.

To do so, issue:

```console
pi64 ~ # echo "net-dialup/freeradius samba" >> /etc/portage/package.use/freeradius
pi64 ~ # emerge -av net-dialup/freeradius
```

PS I don't understand your point about `(/usr)/sbin`? This location is for executables intended for use exclusively by system admins; it does not mandate anything about static or dynamic linking. See e.g. [these notes](https://www.tldp.org/LDP/Linux-Filesystem-Hierarchy/html/sbin.html).

seanuxbridgeperluser, 2019-12-17:
Thanks, that worked!

I could have sworn I had the samba flag set but apparently I did not.

Disregard the sbin comment. In some systems, sbin houses statically linked binaries but this is not that one.

closed 2019-12-17 by seanuxbridgeperluser