sakaki-/gentoo-on-rpi-64bit #131: Kernel recompilation

TheChymera, 2019-12-27:
I would like to recompile the kernel in order to add some custom options, but I notice `/usr/src/linux` is empty. What sources and config did you use for the genpi64lite image?

sakaki-, 2019-12-28:
The kernel used is from [here](https://github.com/sakaki-/bcmrpi3-kernel-bis) (for the RPi3) and [here](https://github.com/sakaki-/bcm2711-kernel-bis) (for the RPi4).
The configs used on the image may be viewed [here](https://github.com/sakaki-/gentoo-on-rpi-64bit/tree/master/configs).
Custom native kernel build instructions in the readme, [here](https://github.com/sakaki-/gentoo-on-rpi-64bit/#optional-compiling-a-kernel-from-source).
Cross-compilation instructions on the project's wiki [here](https://github.com/sakaki-/gentoo-on-rpi-64bit/wiki/Build-an-RPi4-64bit-Kernel-on-your-crossdev-PC) (and [here](https://github.com/sakaki-/gentoo-on-rpi-64bit/wiki/Build-an-RPi3-64bit-Kernel-on-your-crossdev-PC)).

hth, sakaki

TheChymera, 2019-12-28:
Thanks for the info. Relating the options I wanted to enable, I notice the image contains `sys-fs/f2fs-tools`. Are the relevant modules enabled for em to be able to run my root partition as f2fs? The [Gentoo Guide](https://wiki.gentoo.org/wiki/F2FS) says `CONFIG_F2FS_FS_SECURITY` is required, but that appears to not be in [the config](https://github.com/sakaki-/gentoo-on-rpi-64bit/blob/a312992c0a2a92201d39b2bea0fc3be67792ab87/configs/bcmrpi3_defconfig-bis#L5604).

sakaki-, 2020-01-31:
Haven't tried `f2fs` for a while... if you need this setting, please just file a PR ([example](https://github.com/sakaki-/bcm2711-kernel-bis/pull/6)) on the `bis` kernel(s) to turn this on?

TheChymera, 2020-02-04:
While this may become relevant again for kernel recompilation in general, it turns out that for F2FS your kernel “just works”, though sometimes there may be issues (which however are independent of your kernel config). https://github.com/sakaki-/gentoo-on-rpi-64bit/issues/138

closed 2020-02-04 by TheChymera