sakaki-/gentoo-on-rpi-64bit #132: Hardware accelerated videos on youtube?

wof2, 2019-12-28:
Hi, first of all thanks for a hell of a job :)
I'm running Gentoo v1.5.3 on RPI4. I've tried watching 1080p videos on youtube via Chromium and Firefox. The videos are unwatchable - stuttering and CPUs are @ 100%. Is this expected behavior? Shouldn't there be a hardware acceleration running?

sakaki-, 2020-01-02:
Hi @wof2,

unfortunately, although the image now supports access to the h/w video codecs via both V4L2-M2M and MMAL endpoints to access the RPi's built-in hardware video codecs (and ships with a custom version of ffmpeg that can utilise both), neither `firefox` nor `chromium` currently exploit these out of the box (see e.g. [this post](https://www.raspberrypi.org/forums/viewtopic.php?p=1495290#p1495290) and [this thread](https://www.raspberrypi.org/forums/viewtopic.php?f=66&t=233552)), although I think Raspbian's 32-bit custom Chromium build can use MMAL codecs, so it may be possible to adopt some of their patches (not something I've looked at yet). PRs on this topic welcome!

**Hint**: one thing you _can_ do is to play youtube videos using <kbd>Applications</kbd>&rarr;<kbd>Multimedia</kbd>&rarr;<kbd>SMPlayer</kbd>, as this has got acceleration enabled. In this app, you can specify the default quality you want via <kbd>Options</kbd>&rarr;<kbd>Preferences</kbd>, <kbd>Network</kbd> (left tab), and open streams via <kbd>Open</kbd>&rarr;<kbd>URL...</kbd> (just paste in the youtube URL you want to watch).

On my RPi4B (4GB), with a fast IP connection, this plays back 1080p videos smoothly; for instance:

[image]

If you like this route, you can also install `media-video/smtube` as a front end:

```console
demouser@pi64 ~ $ sudo emaint sync --repo genpi64
demouser@pi64 ~ $ sudo emerge -v media-video/smtube
```

Once built, this YouTube browser front-end can be launched by pressing <kbd>F11</kbd> from within SMPlayer.

hth, sakaki

sakaki-, 2020-01-31:
Closing now - please re-open if any issues with the SMPlayer solution above.

Thanks, sakaki

closed 2020-01-31 by sakaki-