sakaki-/gentoo-on-rpi-64bit #135: My screen doesn't display anything after launch:/

Th3darkcatYT, 2019-12-31:
Hello, here is the link of my video which shows my problem (For information, I use a raspberry pi 4B with the 4go of ram . I took the image GenPi64.img.xz) : [https://youtu.be/WUTI_cP5S58](url)

samip5, 2020-01-01:
That looks like a boot loop, does it do that indefinitely, as in unless power is pulled?
If it does, I would recommend trying the lite image, and installing desktop environment by hand. (Gentoo and CLI knowledge required.)

sakaki-, 2020-01-02:
It certainly looks as if it reboots at least once (although doesn't cycle after that). The culprit here is often the power supply - are you using an official RPi unit, or a phone charger or similar?

if you are using an official PSU, try editing the file `/boot/config.txt` (i.e., `config.txt` on the first partition on the microSD card; this is formatted FAT so you can edit it on any PC), and uncommenting the following line, so it now reads:
```ini
hdmi_safe=1
```

Save, leaving the rest of this file as-is, and try rebooting your RPi4B again (the above forces a very basic display; but if it works, you can then tweak further settings to increase the resolution etc.)

If that _also_ fails to display anything, one other thing you can try is to edit the `config.txt` file again and uncomment and change the following line so it reads:
```ini
config_hdmi_boost=9
```
leaving the rest of the file as-is. This will produce a stronger output signal, which is necessary with some displays.

You can find more information about the video settings in `(/boot/)config.txt` [here](https://www.raspberrypi.org/documentation/configuration/config-txt/video.md).

hth, sakaki