sakaki-/gentoo-on-rpi-64bit #136: h265 10bit 4k drop frame

carlsue, 2020-01-03:
the movie is the lion h265 10bit 4k，can't play,maybe soft decode, libreelec is ok.

sakaki-, 2020-01-03:
Hi @carlsue,

Is this streaming over a network, or from a local file? And, which app did you use for playback (generally, SMPlayer has better performance than e.g. VLC)?

best, sakaki

carlsue, 2020-01-06:
> Hi @carlsue,
> 
> Is this streaming over a network, or from a local file? And, which app did you use for playback (generally, SMPlayer has better performance than e.g. VLC)?
> 
> best, sakaki

steaming over 1000M lan,  app is vlc or kodi 18.4(install frome emerge)

sakaki-, 2020-01-06:
VLC is significantly worse than SMPlayer (and kodi won't have h/w acceleration in vanilla form until the next major release, when V4L2-M2M endpoints will be supported).

Accordingly, would you mind trying SMPlayer and let me know if that improves things for you please? It generally has the best performance and does support the RPi's h/w acceleration in 64bits in the current version. This will help narrow down if it is a codec thing (which I suspect it is), or some other issue.

Thx, sakaki

carlsue, 2020-01-07:
> VLC is significantly worse than SMPlayer (and kodi won't have h/w acceleration in vanilla form until the next major release, when V4L2-M2M endpoints will be supported).
>
> Accordingly, would you mind [...]

oooo...  , i sell my pi4 today.