sakaki-/gentoo-on-rpi-64bit #139: How do I install and run mongodb (64-bit) on a raspberrypi 3b+ with this?

AtomynosAtom, 2020-01-21:


sakaki-, 2020-01-21:
The `mongodb` package was recently added to the binhost, so you are in luck (no need to build it locally) ^-^

See my post [here](https://forums.gentoo.org/viewtopic-p-8409482.html#8409482) for further details (will work on an RPi3B/B+/4B).

hth, sakaki

AtomynosAtom, 2020-01-21:
What a coincidence i was just doing that! Thanks anyways for your blazing fast response.

AtomynosAtom, 2020-01-22:
`pi64 ~ # sudo emerge -v dev-db/mongodb                                                                                                                                                                                                                              Local copy of remote index is up-to-date and will be used.                                                                                                                                                                                                          Local copy of remote index is up-to-date and will be used.                                                                                                                                                                                                          These are the packages that would be merged, in order:                                                                                                                                                                                                              Calculating dependencies... done!                                                                                                                                                                                                                                   !!! All ebuilds that could satisfy "dev-db/mongodb" have been masked.                                                             !!! One of the following masked packages is required to complete your request:                                                    - dev-db/mongodb-4.2.2::gentoo (masked by: SSPL-1 license(s), missing keyword)                                                    A copy of the 'SSPL-1' license is located at '/var/db/repos/gentoo/licenses/SSPL-1'.                                                                                                                                                                                - dev-db/mongodb-4.2.1::gentoo (masked by: SSPL-1 license(s), missing keyword)                                                    - dev-db/mongodb-4.0.14::gentoo (masked by: SSPL-1 license(s), missing keyword)                                                   - dev-db/mongodb-4.0.13::gentoo (masked by: SSPL-1 license(s), missing keyword)                                                   - dev-db/mongodb-4.0.12::gentoo (masked by: SSPL-1 license(s), missing keyword)                                                   - dev-db/mongodb-3.6.16::gentoo (masked by: SSPL-1 license(s), missing keyword)                                                   - dev-db/mongodb-3.6.14::gentoo (masked by: SSPL-1 license(s), missing keyword)                                                   - dev-db/mongodb-3.6.13::gentoo (masked by: SSPL-1 license(s), missing keyword)                                                                                                                                                                                     For more information, see the MASKED PACKAGES section in the emerge                                                               man page or refer to the Gentoo Handbook.                                                                                          `

I am not really sure what to do here.

sakaki-, 2020-01-22:
You need first to do:
```console
pi64 ~ # genup
```
and let that run to completion, bringing the rest of your system up to date. May take several hours!

Once done, issue:
```console
pi64 ~ # echo "dev-db/mongodb SSPL-1" > /etc/portage/package.license/mongodb
```

to note that you accept the license terms, then:
```console
pi64 ~ # emerge -v dev-db/mongodb
```

should work.

Best, sakaki