sakaki-/gentoo-on-rpi-64bit #140: why is your firefox on nightly channel?

stefson, 2020-01-25:
hey there, I booted into your image the other day and was surprised that your precompiled firefox binary seems to be on nightly. Why is this so? And where do you share the ebuild?

sakaki-, 2020-01-31:
Hi @stefson,

so, for a while now, the ebuilds used for `firefox` are just the standard Gentoo tree ones, under the `~arm64` arch (issue `eix www-client/firefox` to see). So, it is upstream's choice to use nightly.

You can find the upstream ebuilds [here](https://github.com/gentoo/gentoo/tree/master/www-client/firefox).

hth, sakaki

closed 2020-01-31 by sakaki-

stefson, 2020-05-22:
@sakaki- I finally managed to compile my own binary, and it's not on nightly. Here's what happens if I wanted to replace my own binary with your's instead: 

```
[binary   R   ~] www-client/firefox-76.0.1::gentoo  USE="bindist* hwaccel* lto* pulseaudio* screenshot* startup-notification* system-av1 system-harfbuzz system-icu system-jpeg system-libevent system-libvpx system-webp -clang* -custom-cflags -custom-optimization -debug (-eme-free) -geckodriver -gmp-autoupdate (-hardened) -jack (-pgo) (-selinux) -test -wayland -wifi" CPU_FLAGS_ARM="(-neon)" L10N="en-GB* -ach -af -an -ar -ast -az -be -bg -bn -br -bs -ca -cak -cs -cy -da -de* -dsb -el -en-CA -eo -es-AR -es-CL -es-ES -es-MX -et -eu -fa -ff -fi -fr -fy -ga -gd -gl -gn -gu -he -hi -hr -hsb -hu -hy -ia -id -is -it -ja -ka -kab -kk -km -kn -ko -lij -lt -lv -mk -mr -ms -my -nb -nl -nn -oc -pa -pl -pt-BR -pt-PT -rm -ro -ru -si -sk -sl -son -sq -sr -sv -ta -te -th -tr -uk -ur -uz -vi -xh -zh-CN -zh-TW" 0 KiB

Total: 1 package (1 reinstall, 1 binary), Size of downloads: 0 KiB

```
I wonder if the nightly thing happens due to USE="+bindist" from your binary?