sakaki-/gentoo-on-rpi-64bit #144: Sorry to use this channel, but I need your help

Askmewho, 2020-02-12:
Hi sakaki,ive sent you an email on you published email. I am struggling to get mesa working on a chroot as user, it only works as root. I add the user pi to the video group it worked... But now it doesn't anymore. Close the issue after you red it. And thanks on advance.

gigabyte18, 2020-02-13:
Probably the same problem I have, i was working headless with wifi access point on my laptop wich were connected to a mobile with USB tethered internet from usb. But after a play with quemmu kvm with installed Debian in a container i got some hard reboots (becouse of Firefox installed inside the container was halted).. I cant connect to the access point after boot, I found hdmi monitor connect it and i can see that sistem booted to X now without any connections to internet from wifi, the .sh script in a boot partition not toched.. Wifi after boot process not connected automatically. 
I dont know what to do coz i dont have any keybord to attach it to pi..
The question is how to make wifi connected to my AP after booting here in Gentoo arm64 system, i know how to do it on Ubuntu but Gentoo is a dark hole for me, plz help

gigabyte18, 2020-02-13:
I've tried to chroot into sdcard from m laptop and made genup toonight, nothing changed, monitor with X working when hdmi connected

closed 2020-02-13 by Askmewho