sakaki-/gentoo-on-rpi-64bit #147: MESA-20.0.0 Force using llvmPipe(software rendering)

FuyuriSeiichi, 2020-03-03:
Hi, Sakaki, 

I just do more aggressive update to your image. And after it's now at MESA-20.0.0
While I did ask for VC4 support. But unfortunately, it imposing rendering into llvmpipe according to "glxinfo -B"

Here is the configuration during emerge:
[ebuild   R    ] media-libs/mesa-20.0.0::gentoo  USE="X dri3 egl gallium gbm gles1 gles2 llvm wayland xa xvmc zstd -classic (-d3d9) -debug -libglvnd (-lm-sensors) (-opencl) -osmesa (-selinux) -test -unwind -vaapi (-valgrind) (-vdpau) (-vulkan) (-vulkan-overlay)" VIDEO_CARDS="vc4 -freedreno (-i915) (-i965) (-intel) (-iris) (-lima) -nouveau -panfrost (-r100) (-r200) (-r300) (-r600) (-radeon) (-radeonsi) -virgl -vivante (-vmware)" 11,869 KiB

I wonder that is there a way to revive my VC4 rendering? 
Thank you.

sakaki-, 2020-03-04:
I've [pushed an update](https://github.com/sakaki-/genpi64-overlay/commit/12ff2368b76dccdaedc2d29446960d6267d694c8) for this version of mesa to the overlay. If you update it and use that version, you should find you have h/w acceleration again (provided you have the `rpi-v3d` USE flag active).

The diff wrt the main tree version shows the changes needed to get the h/w acceleration on the Pi4:

```diff
--- /usr/portage/media-libs/mesa/mesa-20.0.0.ebuild	2020-02-19 22:39:40.000000000 +0000
+++ mesa-20.0.0.ebuild	2020-03-04 13:32:33.790204577 +0000
@@ -36,7 +36,7 @@
 
 IUSE="${IUSE_VIDEO_CARDS}
 	+classic d3d9 debug +dri3 +egl +gallium +gbm gles1 +gles2 libglvnd +llvm
-	lm-sensors opencl osmesa selinux test unwind vaapi valgrind vdpau vulkan
+	lm-sensors opencl osmesa rpi-v3d selinux test unwind vaapi valgrind vdpau vulkan
 	vulkan-overlay wayland +X xa xvmc +zstd"
 
 REQUIRED_USE="
@@ -69,6 +69,7 @@
 	video_cards_vmware? ( gallium )
 	xa? ( X )
 	xvmc? ( X )
+	rpi-v3d? ( arm64 video_cards_vc4 )
 "
 
 LIBDRM_DEPSTRING=">=x11-libs/libdrm-2.4.100"
@@ -356,7 +357,7 @@
 		fi
 	fi
 
-	emesonargs+=( -Dplatforms=surfaceless$(use X && echo ",x11")$(use wayland && echo ",wayland")$(use gbm && echo ",drm") )
+	emesonargs+=( -Dplatforms=$(use X && echo "x11,")surfaceless$(use wayland && echo ",wayland")$(use gbm && echo ",drm") )
 
 	if use gallium; then
 		emesonargs+=(
@@ -420,6 +421,8 @@
 		gallium_enable video_cards_lima lima
 		gallium_enable video_cards_panfrost panfrost
 		gallium_enable video_cards_vc4 vc4
+		# enable for Pi4 also
+		use rpi-v3d && gallium_enable video_cards_vc4 v3d
 		gallium_enable video_cards_vivante etnaviv
 		gallium_enable video_cards_vmware svga
 		gallium_enable video_cards_nouveau nouveau
```

FuyuriSeiichi, 2020-03-06:
Hi, Sakaki,
That's so helpful. 
Thank you very much.

closed 2020-03-06 by FuyuriSeiichi