sakaki-/gentoo-on-rpi-64bit #153: Image size?

jonesmz, 2020-04-05:
I notice that the image size is very large compared to the amount of data on the system after installation.

I think you've mentioned in other issues that right now part of the image creation process is manual-by-hand.

Have you considered trying to poke holes in the filesystem, e.g. write all zeros, so that the resulting image is compressed more?

sakaki-, 2020-04-05:
Hello,

`zerofree` is used on the rootfs, and the swap file overwritten with zeros etc., prior to image capture and compression. Unfortunately a compressed filesystem image will never be as efficient as e.g. a compressed tar of the filesystem contents - you can see this by comparing with e.g. the PINN [bootfs + rootfs](https://isshoni.org/pinn/os/gentoo64/) for the same image (about a 10% penalty), which is mechanically constructed from the filesystem via [this script](https://github.com/sakaki-/pinnify). But it is more convenient for most users.

jonesmz, 2020-04-05:
Hrmm.

What do you think about an approach where the root filesystem is shrunk to the left before creating the image file, and then first-boot expands it to the right to fill all available space?