sakaki-/gentoo-on-rpi-64bit #163: pyconfig_gen won't launch under tigervnc session

jpo-, 2020-04-25:
Using rpi as a headless machine with tigervnc.  I've followed the instructions to setup tigervnc, but when I try to launch pyconfig_gen as root I receive the following error:

```
# pyconfig_gen
No protocol specified
qt.qpa.xcb: could not connect to display :1.0
qt.qpa.plugin: Could not load the Qt platform plugin "xcb" in "" even though it was found.
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

Available platform plugins are: minimal, offscreen, xcb.

Aborted
```

jpo-, 2020-04-25:
The issue appears to be that the .Xauthority isn't being exported.  I think that the [VNC wiki page](https://github.com/sakaki-/gentoo-on-rpi-64bit/wiki/Run-a-Remote-VNC-Desktop-on-your-RPi) should be updated to include instructions to export the XAUTHORITY environment variable.  Potentially by adding instructions to include it in the users .bashrc file like this:

`$ echo "export XAUTHORITY=~/.Xauthority" >> .bashrc`

I'm happy to update the wiki page to add the instruction if that is helpful.  Thanks!

sakaki-, 2020-04-25:
Hi @jpo- 

yes, please feel free to update the wiki page, and thanks for the heads up on this point!

Best, sakaki

jpo-, 2020-04-25:
Hi @sakaki- 

Thanks!  I've updated the wiki page.  I need to look into the issue a bit further though as while I can now launch `sudo pyconfig_gen` from the terminal, it still doesn't launch from the xfce settings menu.

Best, jpo-