sakaki-/gentoo-on-rpi-64bit #164: [Feature Request] AUFS + SQUASHFS

Mithrandil, 2020-05-01:
Hello,

I used to make a Gentoo based live distro targeted at netbooks, http://pixielive.org, so faced many of the problems you are facing, same amount of RAM of RPi3 and similarly low speed SD cards and CPU...

Having AUFS + SQUASHFS as kernel modules (and aufs-util) would allow to mount compressed "modules" as filesystem which would dramatically improve some operations due to the slow nature of SD cards.

For example, mounting a compressed module containing /usr/portage as aufs filesystem would make any emerge operation way quicker, orders of magnitude quicked probably

btw, great job creating this gentoo based RPi distro!

Kind regards,
Christian

jonesmz, 2020-09-27:
I am also interested in this functionality.

@Mithrandil do you have any scripts for when you use to make these netbook images?

Mithrandil, 2020-10-03:
> @Mithrandil do you have any scripts for when you use to make these netbook images?

Sure, I will setup a new repository and come back to you.

But please notice gentoo system has changed in 4 years, you will nee to tweek them slightly.

Mithrandil, 2020-10-03:
@jonesmz 

Here it is!
https://github.com/Mithrandil/PixieLive-build-system

the build script are in the Makefiles directory

for the make command please refer to Makefile.x86 or Makefile.amd64 (e.g. make -f Makefile.x86) (I used to make a symlink named Makefile)

Should you need further assistance just ask.