sakaki-/gentoo-on-rpi-64bit #165: libcec is missing and masked as a package. Relevant for Kodi functionality

wdirksen, 2020-05-10:
Love this image so far. Any reason CEC is not installed? This is pretty important for Kodi. I'm new to Gentoo; comfortable in ArchLinux. I thought I could install libcec and possibly rebuild kodi but libcec is 'masked'. Therefore there is a reason it's not included

wdirksen, 2020-05-11:
I see now see that the Gentoo package 'dev-libs/libcec' is not available for arm64.
I wonder why. I just compiled both libcec and libplatform from their git repo's on aarch64 in ArchARM. ArchARM's Rpi4 package is nowhere near as good as this one.

sakaki-, 2020-06-18:
With [this commit](https://github.com/sakaki-/genpi64-overlay/commit/5ae7f980129d5b0923b22ba04e052f5a0c982516) and [this one](https://github.com/sakaki-/genpi64-overlay/commit/555da31a7426d4970795c9521b355ad90c8d6f64), CEC should now (hopefully!) be installed for >=media-tv/kodi-18.7.1.

best, sakaki

sakaki-, 2020-06-25:
[Included](https://github.com/sakaki-/gentoo-on-rpi-64bit/blob/8949302a677d421507adb4c060ee47808fe4c005/reference/world-all-packages#L275) in the v1.6.0 image, which has now been released.

Closing now, please re-open if it doesn't work for you.

Best, sakaki

closed 2020-06-25 by sakaki-