sakaki-/gentoo-on-rpi-64bit #166: wpa2 enterprise cannot connect

dony71, 2020-05-13:
log says below
what wrong?

----------------------------------------------------------------------------------
May 12 17:59:05 pi64 NetworkManager[1820]: <info>  [1589331545.5126] device (wlan0): Activation: starting connection 'OFFICE' (d0f0b692-5282-4262-bf13-3148f240fe93)
May 12 17:59:05 pi64 NetworkManager[1820]: <info>  [1589331545.5131] audit: op="connection-activate" uuid="d0f0b692-5282-4262-bf13-3148f240fe93" name="OFFICE" pid=3322 uid=1001 result="success"
May 12 17:59:05 pi64 NetworkManager[1820]: <info>  [1589331545.5140] device (wlan0): state change: disconnected -> prepare (reason 'none', sys-iface-state: 'managed')
May 12 17:59:06 pi64 NetworkManager[1820]: <info>  [1589331546.0509] device (wlan0): set-hw-addr: reset MAC address to DC:A1:3A:55:0A:87 (preserve)
May 12 17:59:06 pi64 kernel: brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
May 12 17:59:06 pi64 kernel: IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
May 12 17:59:06 pi64 NetworkManager[1820]: <info>  [1589331546.0745] device (wlan0): state change: prepare -> config (reason 'none', sys-iface-state: 'managed')
May 12 17:59:06 pi64 NetworkManager[1820]: <info>  [1589331546.0756] device (wlan0): Activation: (wifi) access point 'OFFICE' has security, but secrets are required.
May 12 17:59:06 pi64 NetworkManager[1820]: <info>  [1589331546.0757] device (wlan0): state change: config -> need-auth (reason 'none', sys-iface-state: 'managed')
May 12 17:59:06 pi64 NetworkManager[1820]: <info>  [1589331546.0813] device (wlan0): state change: need-auth -> prepare (reason 'none', sys-iface-state: 'managed')
May 12 17:59:06 pi64 NetworkManager[1820]: <info>  [1589331546.0824] device (wlan0): state change: prepare -> config (reason 'none', sys-iface-state: 'managed')
May 12 17:59:06 pi64 NetworkManager[1820]: <info>  [1589331546.0833] device (wlan0): Activation: (wifi) connection 'OFFICE' has security, and secrets exist.  No new secrets needed.
May 12 17:59:06 pi64 NetworkManager[1820]: <info>  [1589331546.0834] Config: added 'ssid' value 'OFFICE'
May 12 17:59:06 pi64 NetworkManager[1820]: <info>  [1589331546.0836] Config: added 'scan_ssid' value '1'
May 12 17:59:06 pi64 NetworkManager[1820]: <info>  [1589331546.0836] Config: added 'bgscan' value 'simple:30:-65:300'
May 12 17:59:06 pi64 NetworkManager[1820]: <info>  [1589331546.0836] Config: added 'key_mgmt' value 'WPA-EAP WPA-EAP-SHA256'
May 12 17:59:06 pi64 NetworkManager[1820]: <info>  [1589331546.0837] Config: added 'password' value '<hidden>'
May 12 17:59:06 pi64 NetworkManager[1820]: <info>  [1589331546.0837] Config: added 'eap' value 'PEAP'
May 12 17:59:06 pi64 NetworkManager[1820]: <info>  [1589331546.0837] Config: added 'fragment_size' value '1266'
May 12 17:59:06 pi64 NetworkManager[1820]: <info>  [1589331546.0837] Config: added 'phase2' value 'auth=MSCHAPV2'
May 12 17:59:06 pi64 NetworkManager[1820]: <info>  [1589331546.0838] Config: added 'domain_suffix_match' value 'myoffice.com'
May 12 17:59:06 pi64 NetworkManager[1820]: <info>  [1589331546.0838] Config: added 'identity' value 'guest'
May 12 17:59:06 pi64 NetworkManager[1820]: <info>  [1589331546.0839] Config: added 'proactive_key_caching' value '1'
May 12 17:59:06 pi64 NetworkManager[1820]: <info>  [1589331546.1260] device (wlan0): supplicant interface state: disconnected -> scanning
May 12 17:59:08 pi64 NetworkManager[1820]: <info>  [1589331548.9636] device (wlan0): supplicant interface state: scanning -> associating
May 12 17:59:12 pi64 NetworkManager[1820]: <info>  [1589331552.0110] device (wlan0): supplicant interface state: associating -> associated
May 12 17:59:12 pi64 NetworkManager[1820]: <warn>  [1589331552.0227] sup-iface[0x557b5cb810,wlan0]: connection disconnected (reason -3)
May 12 17:59:12 pi64 NetworkManager[1820]: <info>  [1589331552.0600] device (wlan0): supplicant interface state: associated -> disconnected
May 12 17:59:12 pi64 NetworkManager[1820]: <info>  [1589331552.1263] device (wlan0): supplicant interface state: disconnected -> scanning
May 12 17:59:31 pi64 NetworkManager[1820]: <warn>  [1589331571.4154] device (wlan0): Activation: (wifi) association took too long
May 12 17:59:31 pi64 NetworkManager[1820]: <info>  [1589331571.4155] device (wlan0): state change: config -> need-auth (reason 'none', sys-iface-state: 'managed')
May 12 17:59:31 pi64 NetworkManager[1820]: <warn>  [1589331571.4192] device (wlan0): Activation: (wifi) asking for new secrets
May 12 17:59:31 pi64 dbus-daemon[4209]: [session uid=1001 pid=4207] Activating service name='org.freedesktop.secrets' requested by ':1.24' (uid=1001 pid=4323 comm="nm-applet ")
May 12 17:59:31 pi64 gnome-keyring-daemon[4418]: couldn't access control socket: /run/user/1001/keyring/control: No such file or directory
May 12 17:59:31 pi64 dbus-daemon[4209]: [session uid=1001 pid=4207] Successfully activated service 'org.freedesktop.secrets'
May 12 17:59:31 pi64 gnome-keyring-daemon[4418]: daemon/gkd-main.c:931 Couldn't get object path: GDBus.Error:org.freedesktop.login1.NoSuchSession: No session 'c6' known
May 12 17:59:47 pi64 NetworkManager[1820]: <info>  [1589331587.2209] device (wlan0): supplicant interface state: scanning -> inactive

burnbabyburn, 2020-05-15:
log says above
missing packages

dony71, 2020-05-15:
which package is missing?

youknowwho, 2020-07-16:
@dony71 Hi dony71 I saw this post and I was going to try to help. Are you still having this problem?
So I was looking at this and I don't think I saw mention of missing packages in the log. There is the line "no such file or directory" but I think it's specifically referring to a control socket, which really isn't a real file, so you're probably not missing files (or packages).
When I follow the log, it looks like the device is trying to form an association with your 80211 access point, but then the interface looks like it's being cycled up/down a few times. This is common with wpa_supplicants, but not always easy to spot the reasoning why it could be happening.
The best advice I could probably start with for something like this to look at the /etc/wpa_supplicant/wpa_supplicant.conf configuration for your WPA enterprise configuration. I am thinking that people looking at this issue might be able to spot something in the configuration and at least make sure that the parameters for wpa enterprise look correct, so maybe you could post it and then talk about how your enterprise environment is configured.
For example, I see this line:
May 12 17:59:31 pi64 NetworkManager[1820]: [1589331571.4155] device (wlan0): state change: config -> need-auth (reason 'none', sys-iface-state: 'managed')
This is the interface changing state, and then there's the "need auth" message. So I am thinking the first place to start is the wpa supplicant configuration file.
If you didn't use a wpa_supplicant.conf file to configure the wireless client, maybe you should try that first as a troubleshooting step.
For example, if you used a configuration utility, or networkmanager, it could be possible that something is being abstracted by the configuration utility, and working directly with the wpa_supplicant would expose the problem and then you could implement the change/difference in the config tool (or networkmanager).
Working directly with the supplicant, you could also probably enable some debugging, review the log, and again, maybe see what's going upside down with the client association/wpa authentication. Because WPA enterprise is a little more complex than personal, including many variables in different enterprise environments, like many will like to use RADIUS, certificates, policies with how authentication is managed, the authentication method, and other complex elements that make troubleshooting something like this more complex than just reviewing what we see in the log here.
What do you think? Have you tried working directly with the 802.11 wireless supplicant? You'd probably want to disable networkmanager, configure some debugging temporarily, work on the configuration.