sakaki-/gentoo-on-rpi-64bit #171: Running x86 programs with qemu

rulet, 2020-05-26:
Hello. It's actually not an issue. I have a question if you can help to run x86 programs with qemu on your system on raspberry pi 3 v.B? The purpose is to run simple windows application with wine. Can you provide some instructions for that?

perkinslr, 2020-06-11:
That's really outside the scope of this project...  That said, you need to emerge qemu with `QEMU_USER_TARGETS="x86_64"` (and whatever other targets you want).  You can then set up a chroot with wine in it and run that, similar to how qemu and a chroot can be used to compile arm programs from an x86_64 host.  

All that said, aarch64 (64bit arm) is a fairly RISC instruction set, which can be easily translated to the CISC instruction set of x86_64.  The inverse is *not* true.  Do *not* expect decent performance going this route (i386 dos programs work reasonably inside dosbox, late 90s programs *might* work passably).

sakaki-, 2020-06-11:
Thanks @perkinslr.
Closing.

closed 2020-06-11 by sakaki-