sakaki-/gentoo-on-rpi-64bit #172: Possible to disable binhost functionality?

WAUthethird, 2020-06-02:
Hi,

I'd like to know how to disable the binhost, but retain genup functionality in terms of weekly upgrades.

Thanks.

perkinslr, 2020-06-11:
Remove the line defining the binhost in /etc/portage/make.conf
Note that it will make upgrades take a **long** time.  If you have distcc running in a chroot on a powerful machine, that'll help, but it'll still take a day or more.

sakaki-, 2020-06-11:
Thanks @perkinslr, exactly so. In fact both the below lines can be commented out, thus:

```bash
#FEATURES="${FEATURES} getbinpkg"
#PORTAGE_BINHOST="https://isshoni.org/pi64pie"
```

Closing.

closed 2020-06-11 by sakaki-

WAUthethird, 2020-06-15:
This seems to have worked! Thanks.

I know it's a bit strange of a question to ask, seeing as how slow the Pi4 would be, but I've got a bit of a unique situation, heh.