sakaki-/gentoo-on-rpi-64bit #175: Set LLVM_TARGETS="AArch64 BPF WebAssembly" in make.conf

perkinslr, 2020-06-12:
Just tried updating a fresh copy of the image, it tries to build rust on the pi or inside the chroot instead of hitting the binary package.  This is because the only bin packages available have 
`LLVM_TARGETS="AArch64 BPF WebAssembly"`

set, and the default includes a bunch of un-needed ones (x86_64 and the like).  I recognized what was going on, because I was watching the update since the image is old (3 months), but it would have caused problems if I just left it to do an auto-update in a week.

sakaki-, 2020-06-12:
Hi @perkinslr,

`LLVM_TARGETS` is set, in the [custom profile's `make.defaults`](https://github.com/sakaki-/genpi64-overlay/blob/8a557a1744d3d163609a5b67c8fa8367bc6ebc6f/profiles/targets/rpi3/make.defaults#L62-L63).

I've actually just been running `genup` on vanilla 1.5.4 images over the last few days, as part of the (very) early prep for the next version image release, and `rust` was upgrading fine using binary packages. Did you use `genup` to update, or did you use a different method? Or did you perhaps switch profile?

Best, sakaki

perkinslr, 2020-06-13:
Let's see here... I converted the filesystem to btrfs and applied heavy compression, then I set the locale, keyboard, and timezone correctly.  Then I ran `eix-sync` and `genup`.  That failed, with a message about necessary mask changes, so I re-ran it with `genup -e --autounmask-write`, waited for it to fail again, ran `dispatch-conf` and started it again.  It failed when rust checked for 9+ gb of free space, and I noticed the miss-set flags.  I just ran through that process again from the ext4 version of the image.  

I'll re-download the image again here and see if I can replicate the issue.

perkinslr, 2020-06-13:
Ah, I figured it out.  It seems that I've got environment variables leaking from the host system into the chroot.  Not sure where or why `LLVM_TARGETS` is set on the host, but that seems to be the source of the issue.

closed 2020-06-13 by perkinslr