sakaki-/gentoo-on-rpi-64bit #176: porthole / portage compatibility issues

grbd, 2020-06-19:
Hi,
I noticed with the latest portage sys-apps/portage-2.3.100-r1::gentoo
it doesn't seem to get along with the version of porthole that's installed by default
I don't consider it a big deal or anything since I don't use a gui for updating anyway, so it was just a case of uninstalling porthole to allow portage to update.

Great job on all the work you put into the scripts like genup btw.
I recently added in a wiki entry for switching across to a hard disk
I suspect a lot of the stuff in the Readme could do with being ported across to the wiki (like the network setup)

sakaki-, 2020-06-19:
Thanks @grbd for the wiki entry!

As to porthole, that's gone in the (shortly forthcoming) v1.6.0 image release (and the [updated v1.5.4](https://github.com/sakaki-/genpi64-overlay/commit/b68fae7262e6cd29240a2effcab72d26d42371b3) app set); and is [removed by a fixup](https://github.com/sakaki-/genpi64-overlay/blob/master/app-portage/weekly-genup/files/fixup-0011-depclean-porthole.sh) now for those with it in their existing `/var/lib/portage/world`.

Yes, the docs do need some refactoring - not sure about the right approach there. Smaller chunks in some form of wiki probably make more sense than the current mega-readme. Will have a look once v1.6.0 is out (should be next few days, NMIs permitting).

Best, sakaki

sakaki-, 2020-06-23:
Closing this issue now, as I believe the specific `porthole` point has been addressed.
Best, sakaki

closed 2020-06-23 by sakaki-