sakaki-/gentoo-on-rpi-64bit #177: Non-desktop profile for RPi4

samip5, 2020-06-20:
Could you make a profile that's non-desktop? There seems to be only desktop versions when doing eselect profile list, unless you count the Gentoo default ones.

necrose99, 2020-08-29:
@samip5  he has a minimal image of which you can add xfce back , however this is nice if you have [moba-xterm](https://mobaxterm.mobatek.net/) or alike , 

leafpad edit my-config.cfg  however x11/xfce will not run by default ie start on demand... which is good , ie config apache or nginx  php 
has multiple use expander's ...   which if adding features ie brolti compression....  or truetype to php can be a touch thornier 

and more of a gui editor can be useful to brek to 4-5 ngnix lines.. etc..

samip5, 2020-08-29:
Who?

closed 2020-10-17 by samip5