sakaki-/gentoo-on-rpi-64bit #178: sysctl.conf parameters not being set at boot time

youknowwho, 2020-06-25:
Hello,
Thank you so much for this project! This is very exciting work, thank you for this work.
I did have a question about something I am working on with the new image. This might already be a bug or a known issue in the Gentoo GNU/Linux software, but I wasn't able to find anything. If we think it's a gentoo bug (and not something as part of this image) please let me know and I will try to replicate this using x86_64 or another arch and then I will file a bug report. But I have searched and searched and cannot seem to find anything, I am hoping you will be able to help me find out what's going wrong.
I am using the "genpi64.img" image, md5 hash is 
After installing the image on a raspberry pi 4, I am trying to make a couple of changes to the /etc/sysctl.conf file, and wanting to see those parameters get set at boot time. But these are not getting set.
For example, if I set these in the etc/sysctl.conf:
#######################################
cat /etc/sysctl.conf  | grep -v \#
net.ipv4.ip_forward = 0
net.ipv6.conf.default.use_tempaddr = 2
net.ipv6.conf.all.use_tempaddr = 2
net.ipv6.conf.all.accept_ra_rt_info_max_plen=64
net.ipv6.conf.default.accept_ra_rt_info_max_plen=64
#######################################

After a fresh boot, I would expect to see the at /proc/sys/net/ipv6/conf/all/use_tempaddr set to 2, but it's set to zero.
cat /proc/sys/net/ipv6/conf/all/use_tempaddr
0

cat /proc/sys/net/ipv6/conf/all/accept_ra_rt_info_max_plen
0

the sysctl init process seems to be running at boot time:
sudo rc-status boot | grep sysc
 sysctl                                                            [  started  ]

If I manually invoke the sysctl process, the values get properly set in the kernel.
sudo sysctl -p
net.ipv4.ip_forward = 0
net.ipv6.conf.default.use_tempaddr = 2
net.ipv6.conf.all.use_tempaddr = 2
net.ipv6.conf.all.accept_ra_rt_info_max_plen = 64
net.ipv6.conf.default.accept_ra_rt_info_max_plen = 64

cat /proc/sys/net/ipv6/conf/default/use_tempaddr 
2

at /proc/sys/net/ipv6/conf/all/use_tempaddr
2

cat /proc/sys/net/ipv6/conf/all/accept_ra_rt_info_max_plen
64

Again, after rebooting the Pi4, the values are reset back to their defaullts again.

Is this a bug, or is the sysctl process somehow not working?

Thank you again for your thoughts and ideas about it! Thank you again for working on this project!

Best Regards,
YKW

sakaki-, 2020-06-26:
Hi @youknowwho,

the recommended way on Gentoo is to put your custom `sysctl` overrides in `/etc/sysctl.d/<rulename>.conf`.
So, for example, if you put, into `/etc/sysctl.d/99-local.conf`:

```ini
vm.vfs_cache_pressure = 51
net.ipv6.conf.default.use_tempaddr = 2
```

save, and reboot, you should find that e.g.:

```console
demouser@pi64 ~ $ cat /proc/sys/vm/vfs_cache_pressure
51
```

So far so good - looks like our settings were taken. **However**:

```console
demouser@pi64 ~ $ cat /proc/sys/net/ipv6/conf/default/use_tempaddr
0
```

What gives? Well, it's because in the current kernel (following upstream's configuration), `ipv6` [**is a module**](https://github.com/sakaki-/gentoo-on-rpi-64bit/blob/caa93830a3a11eb6d68a2053f5d3900e494bb6e2/configs/bcm2711_defconfig-bis#L897), and not yet loaded at the (very early) point `sysctl` is called. So the `sysctl` settings targetted at it are ignored.

To workaround this, edit `/etc/conf.d/modules`, *adding* the value to any existing uncommented entry (or making a new entry if none exists). For example, on the v1.6.0 image you'd change the final line of this file so it read:

```ini
modules_5_4="snd_bcm2835 ipv6"
```

> This assumes you are using a 5.4 kernel, which already had the `snd_bcm2835` module autoloaded. Adapt accordingly.

Save the file and restart. You should now find that:

```console
demouser@pi64 ~ $ cat /proc/sys/net/ipv6/conf/default/use_tempaddr
2
```
yay!

One last note of warning though: when `NetworkManager` comes up (later in the boot sequence) and activates a connection, it _overrides_ a number of these (network-related) `sysctl` settings. See e.g. [these notes](https://wiki.archlinux.org/index.php/IPv6#NetworkManager) and [this caution](https://access.redhat.com/documentation/en-us/red_hat_enterprise_linux/8/html/configuring_and_managing_networking/disabling-ipv6-on-a-system-that-uses-networkmanager_configuring-and-managing-networking):

> The NetworkManager service sets certain sysctl values when it starts a connection. 

Double-check that doesn't apply to any of the settings you require.

hth, sakaki

youknowwho, 2020-06-26:
Hello sakaki,
Thank you for the fast response, and super thank you so much for that explanation! I appreciated learning the recommended way to configure the sysctl overrides for Gentoo and grateful you shared. When I was searching, I did find some documentation explaining this, and like you said, I also tried that, and it also didn't produce the expected results, but you have helped me understand the reasoning for that issue perfectly: the fact that ipv6 is a module, and not compiled in. It completely makes sense now.
I followed your instructions and everything worked perfectly, exactly as you mentioned, just to recap:
1. I configured the custom override in /etc/sysctl.d/<rulename>.conf.
2. I configured the /etc/conf.d/modules file so the ipv6 module gets loaded at the correct time.
3. Restart, and like you said, yay, it worked, everything worked perfect.
Also, thank you for the warning on the NetworkManager behavior. I know this is probably deprecated also, but I am using /etc/conf.d/net and /etc/init.d/net.lo (and making symbolic links for /etc/init.d/net.lo) and thus, I went ahead and disabled the NetworkManager service, so I am expecting the overrides to still be preserved (until net.lo gets gets deprecated or removed completely or something like that).
I was going to ask another question please, for my own knowledge: If I recompiled this kernel and changed the config so that I made ipv6 built-in, would the workaround (to specify ipv6 in the /etc/conf.d/modules file) no longer be needed? I know this is probably inconsistent when compared to the upstream configuration, I get it, but I was just curious if you think it'd work.
Super thanks again for this project, this is so great!

sakaki-, 2020-06-26:
Hi @youknowwho ,

yes, exactly, if you compiled `ipv6` into the kernel (rather than as a module), your custom `sysctl` settings applying to it (e.g. `net.ipv6.conf.default.use_tempaddr = 2`) would have worked (just like the `vm.vfs_cache_pressure` one did in the example above).

hth, sakaki

youknowwho, 2020-06-27:
@sakaki- Thank you so much again for that explanation, everything is working great. Thanks again for the project, I really admire this work and appreciate your time with this again.

closed 2020-06-27 by youknowwho

youknowwho, 2020-07-02:
Hi Folks,
As a follow through, I did re-compile the kernel with IPv6 built in and everything worked as expected. No longer do I need to specify it in the /etc/conf.d/modules file (now that it is compiled in, not compiled as a module).
Thank you again for  your help!