sakaki-/gentoo-on-rpi-64bit #179: ponyc , ghc

necrose99, 2020-06-25:
https://data.gpo.zugaina.org/stefantalpalaru/dev-lang/pony/ 
newer , 
https://github.com/ponylang/corral ie like cargo... but no for ebuilds atm or ponyup... 

https://data.gpo.zugaina.org/haskell/dev-lang/ghc/ghc-8.8.2.ebuild  
then can build more newer.. if i had a server or distcc ,  on other rpi's. /rock64 ... 
however i got bit greedy n  pushed rpi4 to the **uncle point** _plz no... more emerge foo...._ 
5/7 and near 90% died .... on newer gentoo ghc bootstrap/makebinary ... etc.. 
https://www.dropbox.com/sh/gfsfqv3p4lvnzi6/AAAE-Mb_8ko1oX3xSROVkP8ia?dl=0 ghc built.. 

will run , cabal atm ..... few upstream bins ie emerge cabal-bin , emerge cabal .... 
if you can find a recent cabal... bin.. 

I'd warn less you have 256 gig card or larger ssd /drive over usb ... to avoid , GHC 8 gigs/2gigs ... useful but a 🐖 🐷 

256 gig , long card in rpi4 ,  
https://github.com/necrose99/gentoo-binhost few goodies like snes9x or the like.  few items ie @pentoo 
grub+uboot+aarch64-efi .... 

however they pythonic libs/script  can break between host nodes so uploads can break-ish. .. , had to add a .gitinore so .py  wont upload on gh.. /etc/portage... mirror. 
api key swap... n pray... 
 setting up a more expansive binhost ... as anything larger than 25 megs ..... tends to not upload.  

rock64-pro if i get anm.2  for it , of the 1tb variety prolly can have rpi4-gentoo-docker images , build/push. 
emerge really-bad-idea/unstable ?? in docker if it packages to host test... remove... 
 grub2 is built to the 9's with uboot+efi

tiborh, 2020-07-27:
Extra info about ghc:
(as requested by the failing install:)
$ sudo emerge dev-lang/ghc

Local copy of remote index is up-to-date and will be used.

Local copy of remote index is up-to-date and will be used.
Calculating dependencies... done!
>>> Verifying ebuild manifests
>>> Running pre-merge checks for dev-lang/ghc-8.0.2
 * Checking for at least 8 GiB disk space at "/var/tmp/portage/dev-lang/ghc-8.0.2/temp" ...                                                                                           [ ok ]
 * Checking for at least 2 GiB disk space at "/usr" ...                                                                                                                               [ ok ]
>>> Emerging (1 of 2) dev-lang/ghc-8.0.2::gentoo
>>> Failed to emerge dev-lang/ghc-8.0.2, Log file:
>>>  '/var/tmp/portage/dev-lang/ghc-8.0.2/temp/build.log'
>>> Jobs: 0 of 2 complete, 1 failed                 Load avg: 3.23, 2.84, 2.89
 * Package:    dev-lang/ghc-8.0.2
 * Repository: gentoo
 * Maintainer: haskell@gentoo.org
 * USE:        arm64 elibc_glibc gmp kernel_linux profile userland_GNU
 * FEATURES:   network-sandbox preserve-libs sandbox userpriv usersandbox
 * Checking for at least 8 GiB disk space at "/var/tmp/portage/dev-lang/ghc-8.0.2/temp" ...
 [ ok ]
 * Checking for at least 2 GiB disk space at "/usr" ...
 [ ok ]
>>> Unpacking source...
>>> Unpacking ghc-8.0.2-src.tar.xz to /var/tmp/portage/dev-lang/ghc-8.0.2/work
>>> Unpacking ghc-bin-8.0.2-arm64.tbz2 to /var/tmp/portage/dev-lang/ghc-8.0.2/work

bzip2: /var/tmp/portage/dev-lang/ghc-8.0.2/distdir/ghc-bin-8.0.2-arm64.tbz2: trailing garbage after EOF ignored
>>> Source unpacked in /var/tmp/portage/dev-lang/ghc-8.0.2/work
>>> Preparing source in /var/tmp/portage/dev-lang/ghc-8.0.2/work/ghc-8.0.2 ...
 * strip-flags: CFLAGS: changed '-march=armv8-a+crc -mtune=cortex-a72 -ftree-vectorize -O2 -pipe' to '-march=armv8-a+crc -mtune=cortex-a72 -O2 -pipe'
 * strip-flags: CXXFLAGS: changed '-march=armv8-a+crc -mtune=cortex-a72 -ftree-vectorize -O2 -pipe' to '-march=armv8-a+crc -mtune=cortex-a72 -O2 -pipe'
 * Relocating 'lib' to 'lib64' (bug #476998)
/var/tmp/portage/dev-lang/ghc-8.0.2/temp/ghc-pkg-8.0.2-orig: line 11: /var/tmp/portage/dev-lang/ghc-8.0.2/work/usr/lib64/ghc-8.0.2/bin/ghc-pkg: No such file or directory
 * ERROR: dev-lang/ghc-8.0.2::gentoo failed (prepare phase):
 *   failed to update cache after relocation
 * 
 * Call stack:
 *     ebuild.sh, line  125:  Called src_prepare
 *   environment, line 3723:  Called relocate_ghc '/var/tmp/portage/dev-lang/ghc-8.0.2/work'
 *   environment, line 3484:  Called die
 * The specific snippet of code:
 *       "$gp_back" recache || die "failed to update cache after relocation";
 * 
 * If you need support, post the output of `emerge --info '=dev-lang/ghc-8.0.2::gentoo'`,
 * the complete build log and the output of `emerge -pqv '=dev-lang/ghc-8.0.2::gentoo'`.
 * The complete build log is located at '/var/tmp/portage/dev-lang/ghc-8.0.2/temp/build.log'.
 * The ebuild environment file is located at '/var/tmp/portage/dev-lang/ghc-8.0.2/temp/environment'.
 * Working directory: '/var/tmp/portage/dev-lang/ghc-8.0.2/work/ghc-8.0.2'
 * S: '/var/tmp/portage/dev-lang/ghc-8.0.2/work/ghc-8.0.2'

 * Messages for package dev-lang/ghc-8.0.2:

 * ERROR: dev-lang/ghc-8.0.2::gentoo failed (prepare phase):
 *   failed to update cache after relocation
 * 
 * Call stack:
 *     ebuild.sh, line  125:  Called src_prepare
 *   environment, line 3723:  Called relocate_ghc '/var/tmp/portage/dev-lang/ghc-8.0.2/work'
 *   environment, line 3484:  Called die
 * The specific snippet of code:
 *       "$gp_back" recache || die "failed to update cache after relocation";
 * 
 * If you need support, post the output of `emerge --info '=dev-lang/ghc-8.0.2::gentoo'`,
 * the complete build log and the output of `emerge -pqv '=dev-lang/ghc-8.0.2::gentoo'`.
 * The complete build log is located at '/var/tmp/portage/dev-lang/ghc-8.0.2/temp/build.log'.
 * The ebuild environment file is located at '/var/tmp/portage/dev-lang/ghc-8.0.2/temp/environment'.
 * Working directory: '/var/tmp/portage/dev-lang/ghc-8.0.2/work/ghc-8.0.2'
[... 108 lines cut ...]
bzip2: /var/tmp/portage/dev-lang/ghc-8.0.2/distdir/ghc-bin-8.0.2-arm64.tbz2: trailing garbage after EOF ignored
>>> Source unpacked in /var/tmp/portage/dev-lang/ghc-8.0.2/work
>>> Preparing source in /var/tmp/portage/dev-lang/ghc-8.0.2/work/ghc-8.0.2 ...
 * strip-flags: CFLAGS: changed '-march=armv8-a+crc -mtune=cortex-a72 -ftree-vectorize -O2 -pipe' to '-march=armv8-a+crc -mtune=cortex-a72 -O2 -pipe'
 * strip-flags: CXXFLAGS: changed '-march=armv8-a+crc -mtune=cortex-a72 -ftree-vectorize -O2 -pipe' to '-march=armv8-a+crc -mtune=cortex-a72 -O2 -pipe'
 * Relocating 'lib' to 'lib64' (bug #476998)
/var/tmp/portage/dev-lang/ghc-8.0.2/temp/ghc-pkg-8.0.2-orig: line 11: /var/tmp/portage/dev-lang/ghc-8.0.2/work/usr/lib64/ghc-8.0.2/bin/ghc-pkg: No such file or directory
 * ERROR: dev-lang/ghc-8.0.2::gentoo failed (prepare phase):
 *   failed to update cache after relocation
 * 
 * Call stack:
 *     ebuild.sh, line  125:  Called src_prepare
 *   environment, line 3723:  Called relocate_ghc '/var/tmp/portage/dev-lang/ghc-8.0.2/work'
 *   environment, line 3484:  Called die
 * The specific snippet of code:
 *       "$gp_back" recache || die "failed to update cache after relocation";
 * 
 * If you need support, post the output of `emerge --info '=dev-lang/ghc-8.0.2::gentoo'`,
[... cut ...] qv '=dev-lang/ghc-8.0.2::gentoo'`.
 * The complete build log is located at '/var/tmp/portage/dev-lang/ghc-8.0.2/temp/build.log'.
 * The ebuild environment file is located at '/var/tmp/portage/dev-lang/ghc-8.0.2/temp/environment'.
 * Working directory: '/var/tmp/portage/dev-lang/ghc-8.0.2/work/ghc-8.0.2'
 * S: '/var/tmp/portage/dev-lang/ghc-8.0.2/work/ghc-8.0.2'

[Link to /var/tmp/portage/dev-lang/ghc-8.0.2/temp/environment](https://github.com/tiborh/logs/blob/master/ghc_build_environment.tar.gz)

necrose99, 2020-07-27:
@tiborh Haskell overlay ,  **BINARY**   
_arch_binaries="$arch_binaries arm64? ( https://slyfox.uni.cx/~slyfox/distfiles/ghc-bin-${PV}-aarch64-unknown-linux-gnu.tbz2 )"_
https://data.gpo.zugaina.org/haskell/dev-lang/ghc/

this at least bootstraps system up, 
https://github.com/necrose99/gentoo-binhost/releases/tag/aarch64-coretexa53-linux-gnu%2Fdev-haskell

tiborh, 2020-10-10:
It is now that I got around to this problem. (progressing one problem at a time.) (I'm too new to Gentoo to fully understand how to interpret @necrose99's advice, so will need some time to see if I can use it.
* Anyhow, the first two links are already dead (404 errors).
* Starting with from the github link and attempting an install following the description there.