sakaki-/gentoo-on-rpi-64bit #180: One line change to meta package for systemd / openrc rebuild

grbd, 2020-06-25:
Hi,
I've recently gotten systemd working and I'm about to add something to the wiki for it (with a big experimental warning)

## rpi-64bit-meta

I needed to add a one liner change to the meta package

From
```
>=sys-fs/eudev-3.2.9
```

To
```
!systemd? ( >=sys-fs/eudev-3.2.9 )
```

I think someone else has already put in a pull request for it

## openrc

Part of the process involves rebuilding openrc without netifrc.
However the mainline has switched over to gcc-10 recently, which needs a patch to compile
```
PATCHES=("${FILESDIR}"/${PN}-0.42.1-gcc-10.patch)
```
This might be something for consideration for the next version, since I think your using a custom one in the overlay

sakaki-, 2020-06-25:
Ack. Got to catch up on some real-world NMIs now after the push to get v1.6.0 released burned my last few days, but will address the above thereafter.
Thx, S.

grbd, 2020-06-25:
cool thanks

sakaki-, 2020-06-30:
OK, hopefully these two points are addressed now, via https://github.com/sakaki-/genpi64-overlay/commit/4600d056072fc3a9d6f4899e90764159fc2856db and https://github.com/sakaki-/genpi64-overlay/commit/7a483b4fd459e62f2395951b26969025bb932194.

Please reopen if not the case,

best,
S.

closed 2020-06-30 by sakaki-