sakaki-/gentoo-on-rpi-64bit #182: RE: about swap wiki entry.

k0nsl, 2020-07-10:
Hello sakaki,

I am currently testing `gentoo-on-rpi-64bit` and so far I like what I see. Very good. There's just one thing I accidentally stumbled across, the wiki entry about swap!

The way I see it: I think that introducing an “old school” type of swap file on any RPi-device is a very bad idea due to the nature of swap; **it will tear away the lifespan of your micro SD card or USB flash drive significantly and fast**. If you must use swap, then look into something called ZRAM — this is faster and more safe than the method you describe in the wiki entry about swap.

ZRAM was released sometime in 2014, IIRC. In any case, heed my advice and use ZRAM, at least if you care about the health of your micro SD card or USB flash drive. 👌

That's all. Keep up the good work!

sakaki-, 2020-07-11:
Thanks. The image [does use](https://github.com/sakaki-/genpi64-overlay/blob/872558f6bb18737b372eee84deb1fcb366763846/sys-apps/rpi3-zswap/files/init.d_rpi3-zswap-2) [`zswap`](https://askubuntu.com/questions/471912/zram-vs-zswap-vs-zcache-ultimate-guide-when-to-use-which-one/472227#472227) to attempt to minimize backing store I/O.

Best, sakaki

k0nsl, 2020-07-11:
Aha, I see. I hadn't seen that. Very nice!

Have a great day, sakaki. 😁

closed 2020-07-11 by k0nsl