sakaki-/gentoo-on-rpi-64bit #183: gpg: Bad signature

PatrickLohan, 2020-07-12:
IWhen verifying with gpg I get back the Bad signature error.

Steps tried:
```
wget -c https://github.com/sakaki-/gentoo-on-rpi-64bit/releases/download/v1.6.0/genpi64.img.xz
wget -c https://github.com/sakaki-/gentoo-on-rpi-64bit/releases/download/v1.6.0/genpi64.img.xz.asc
gpg --keyserver hkp://ipv4.pool.sks-keyservers.net:80 --recv-key DDE76CEA
gpg --verify genpi64.img.xz.asc genpi64.img.xz
```
I also tried the direct link for the same(?) file.

sakaki-, 2020-07-12:
Please post the output of:
```console
$ md5sum genpi64.img.xz.asc genpi64.img.xz
```

PatrickLohan, 2020-07-12:
```
b2c6163c357f7d37a162b6bb4a9d0872  genpi64.img.xz.asc
12b78242d86373c9479d45af58b419c6  genpi64.img.xz
```

PatrickLohan, 2020-07-12:
I did try downloading several times. I can try again this evening just in case.

sakaki-, 2020-07-12:
Your `genpi64.img` file is correct, but something is wrong with your copy of the signature file, `genpi64.img.xz.asc`. The `md5sum` output _should_ be:
```
7f331efc323ddd9442d4fdf374fddd41  genpi64.img.xz.asc
12b78242d86373c9479d45af58b419c6  genpi64.img.xz
```

I just downloaded the two files from GitHub again myself using `wget`, to check, and everything looks ok.

Delete `genpi64.img.xz.asc` and try fetching it again.

hth, sakaki

sakaki-, 2020-07-12:
Here's the contents of `genpi64.img.xz.asc` itself:
```
-----BEGIN PGP SIGNATURE-----

iQIzBAABCAAdFiEEQSCyFISndkdGWDkJMcgNst3nbOoFAl7ylM8ACgkQMcgNst3n
bOrThA/7BQ47flVCqJ/qZI+bd6TUSthKVYthGOwrDxmhIcmWXaXYrWTayxpRprHc
JW/AiX7Y3kIK2XmsY0IDZ6Hg+U++3Gz/59Hi7Tv8Fctz5rtUa+gZdPzamUCJfkSR
dNcguowTjMCM6IlQKNP7hbQSoyw3HyUjxel5B7YY9MqNAKVJ+BYNfxL7S2BlvS+r
tTm0tRS7KZrL1ueW0/buy8UOWjVi1XiagtLEV8ibSqdF49FCdiYjh/zeSIYgv/Te
G/fp3ZIc2Xu8bXtSlkx83Ky7tEqEy41PPhwhpAcOiIe8yRf/WOwIF1ySUOPUypa5
tge9XwINgkLCkoqmtDjpJCen9RYHbzl31vQKpTgfjvg3UjgxZ199UmDqYjomnsCo
iCAbk4AAVuwyEOlsoBpc8Esh9d9tp8Gr/Mqw3CJPk5UANatozaOwJOgaYIHMF4hu
V9Y7Jzz5D2zK6A/jFUWPRP+/xEGLpkoIBm7MOkz4NMGLvSf/jP35cIJnd3rB/eMk
oJtJOeDdwp2CUDOqs0/OE64knj936rByZSXWrSyJkI2zxM/1eJTEDviSzqE9dx/1
Y+jgQRaL6pHJYmfvEmkGc3/R4amVmSxFuHCCnVH8BoNqMC+F5PNt2NlEoiZAC4IV
y2BzBu+2PRs5X6R44WzFMNkVadMITNBeGYOWypr1Du9uSo+aicw=
=9uJO
-----END PGP SIGNATURE-----
```

PatrickLohan, 2020-07-12:
Well, basic error, thanks for advising to dl the asc file again! That was indee where the error was, and now I've got the usual 'Good sig, not certified etc.' message. Thanks @sakaki-

closed 2020-07-12 by PatrickLohan