sakaki-/gentoo-on-rpi-64bit #185: Putting /tmp/, /run/, and /var/tmp/ on zram?

jonesmz, 2020-07-16:
I needed to install a package that wasn't available in a binhost, and it took over 12 hours to compile and run tests. I think a large amount of the problem was disk access.

Thoughts on mounting /tmp/, /run/, and /var/tmp/ on zram block devices?

AlexMikhalev, 2020-08-03:
@jonesmz do you have example configuration described in blog or doc? I am keen to implement at least as the script.

jonesmz, 2020-08-03:
I just installed the 'zram-init' ebuild, and then ran 
systemctl enable --now zram_swap zram_tmp zram_var_tmp

You're also supposed to edit /boot/cmdline.txt and /etc/modprob.d/zram.conf, but i'm not sure how necessary that is.

AlexMikhalev, 2020-08-03:
Seems like zswap already part of the image: https://github.com/sakaki-/genpi64-overlay/tree/master/sys-apps/rpi3-zswap
Did you notice any performance improvements after enabling zram init?

jonesmz, 2020-08-03:
zswap is not the same as zram.

Yes, enabling zram_var_tmp speeds up small package compilations enormously.

AlexMikhalev, 2020-08-05:
Looking at write up [here](https://askubuntu.com/questions/471912/zram-vs-zswap-vs-zcache-ultimate-guide-when-to-use-which-one/472227#472227) zswap seems to be more appropriate for Rpi 3/4. 
Have you tried zram init vs putting /var/tmp/ on tmpfs?

jonesmz, 2020-08-05:
Why is compressed in-memory "swap" more appropriate for rpi3/4 for purposes of reducing disk access to /var/tmp/ while emerging packages?

I used zram via zram-init for /var/tmp/. I did not bother with tmpfs because the rpi3 and rpi4 are memory constrained.