sakaki-/gentoo-on-rpi-64bit #187: V4L2 decoding via docker

piaste, 2020-07-26:
Hello. First, thank you a _lot_ for this image and even more for its excellent documentation.

I'm trying to enable Jellyfin hardware acceleration using the [LinuxServer.io image](https://hub.docker.com/r/linuxserver/jellyfin) on a Pi4 4GB, with the 1.6.0 lite image.

Jellyfin has access to the following relevant paths:

```
       - "/opt/vc/lib:/opt/vc/lib"
        - /dev/vc-mem:/dev/vc-mem
        - /dev/vcsm-cma:/dev/vcsm-cma # the documentation says /dev/vcsm
        - /dev/vchiq:/dev/vchiq
        # these 3 devices should be the only one required for V4L2
        - /dev/video10:/dev/video10
        - /dev/video11:/dev/video11
        - /dev/video12:/dev/video12
```

and `/boot/config.txt` has the following settings enabled (unchanged from default except for extra `gpu_mem`):

```
dtparam=audio=on
dtoverlay=vc4-fkms-v3d
gpu_mem=320
start_x=1
```

The two available options are OpenMAX (with MMAL for h.264 decoding) and V4L2. OpenMAX isn't available on 64bit, as the project readme says, and indeed the playback crashes because the `bcm_host` library is missing from `/opt/vc/lib`... although it **is** present in `/usr/lib64`, which has me a little confused.

Switching to V4L2 gives this error when I try to play a HEVC file:

> Stream mapping:
>   Stream #0:0 -> #0:0 (hevc (native) -> h264 (h264_v4l2m2m))
>   Stream #0:1 -> #0:1 (aac (native) -> mp3 (libmp3lame))
> Press [q] to stop, [?] for help
> [h264_v4l2m2m @ 0x55a982a7c0] Using device /dev/video11
> [h264_v4l2m2m @ 0x55a982a7c0] driver 'bcm2835-codec' on card 'bcm2835-codec-encode' in mplane mode
> [h264_v4l2m2m @ 0x55a982a7c0] requesting formats: output=YU12 capture=H264
> [h264_v4l2m2m @ 0x55a982a7c0] Encoder requires yuv420p pixel format.
> Error initializing output stream 0:0 -- Error while opening encoder for output stream #0:0 - maybe incorrect parameters such as bit_rate, rate, width or height

Googling this error isn't leading me anywhere as `yuv420p` seems to be a default ffmpeg codec..

After reading a bunch of threads, I found [this one](https://www.raspberrypi.org/forums/viewtopic.php?t=234778) where @sakaki- was investigating V4L2 support on 64-bit. So I tried the same commands in one of the posts, and the V4L2-powered transcoding appeared to work properly:

```
> wget -c http://mirrors.standaloneinstaller.com/video-sample/grb_2.webm
> time ffmpeg -i grb_2.webm -vcodec h264_v4l2m2m grb_2-v4l2.mov
ffmpeg version 4.2.3 Copyright (c) 2000-2020 the FFmpeg developers
  built with gcc 10.1.0 (Gentoo 10.1.0 p1)
  configuration: --prefix=/usr --libdir=/usr/lib64 --shlibdir=/usr/lib64 --docdir=/usr/share/doc/ffmpeg-4.2.3-r1/html --mandir=/usr/share/man --enable-shared --cc=aarch64-unknown-linux-gnu-gcc --cxx=aarch64-unknown-linux-gnu-g++ --ar=aarch64-unknown-linux-gnu-ar --optflags='-march=armv8-a+crc -mtune=cortex-a72 -ftree-vectorize -O2 -pipe' --disable-static --enable-avfilter --enable-avresample --disable-stripping --disable-optimizations --disable-libcelt --enable-nonfree --disable-indev=oss --disable-indev=jack --disable-outdev=oss --enable-bzlib --disable-runtime-cpudetect --disable-debug --disable-gcrypt --disable-gnutls --disable-gmp --enable-gpl --disable-hardcoded-tables --enable-iconv --disable-libtls --disable-libxml2 --disable-lzma --enable-network --disable-opencl --enable-openssl --enable-postproc --disable-libsmbclient --enable-ffplay --enable-sdl2 --disable-vaapi --disable-vdpau --enable-xlib --enable-libxcb --enable-libxcb-shm --enable-libxcb-xfixes --enable-zlib --disable-libcdio --disable-libiec61883 --disable-libdc1394 --disable-libcaca --disable-openal --enable-opengl --enable-libv4l2 --enable-libpulse --enable-libdrm --disable-libjack --disable-libopencore-amrwb --disable-libopencore-amrnb --disable-libcodec2 --disable-libdav1d --disable-libfdk-aac --disable-libopenjpeg --disable-libbluray --disable-libgme --disable-libgsm --disable-libaribb24 --enable-mmal --disable-libmodplug --disable-libopus --disable-libilbc --disable-librtmp --disable-libssh --disable-libspeex --disable-libsrt --enable-librsvg --disable-ffnvcodec --enable-libvorbis --disable-libvpx --disable-libzvbi --disable-appkit --disable-libbs2b --disable-chromaprint --disable-cuda-llvm --disable-libflite --disable-frei0r --disable-libfribidi --disable-fontconfig --disable-ladspa --disable-libass --disable-libtesseract --disable-lv2 --enable-libfreetype --disable-libvidstab --disable-librubberband --disable-libzmq --disable-libzimg --disable-libsoxr --enable-pthreads --disable-libvo-amrwbenc --enable-libmp3lame --disable-libkvazaar --disable-libaom --disable-libopenh264 --disable-libsnappy --enable-libtheora --disable-libtwolame --disable-libwavpack --disable-libwebp --enable-libx264 --disable-libx265 --enable-libxvid --disable-gnutls --disable-mipsdsp --disable-mipsdspr2 --disable-mipsfpu --disable-altivec --disable-amd3dnow --disable-amd3dnowext --disable-aesni --disable-avx --disable-avx2 --disable-fma3 --disable-fma4 --disable-mmx --disable-mmxext --disable-sse --disable-sse2 --disable-sse3 --disable-ssse3 --disable-sse4 --disable-sse42 --disable-xop --cpu=armv8-a+crc --disable-doc --disable-htmlpages --enable-manpages
  libavutil      56. 31.100 / 56. 31.100
  libavcodec     58. 54.100 / 58. 54.100
  libavformat    58. 29.100 / 58. 29.100
  libavdevice    58.  8.100 / 58.  8.100
  libavfilter     7. 57.100 /  7. 57.100
  libavresample   4.  0.  0 /  4.  0.  0
  libswscale      5.  5.100 /  5.  5.100
  libswresample   3.  5.100 /  3.  5.100
  libpostproc    55.  5.100 / 55.  5.100
Input #0, matroska,webm, from 'grb_2.webm':
  Metadata:
    encoder         : Lavf57.66.105
  Duration: 00:00:27.89, start: 0.033000, bitrate: 157 kb/s
    Stream #0:0: Video: vp9 (Profile 0), yuv420p(tv), 720x480, SAR 8:9 DAR 4:3, 29.97 fps, 29.97 tbr, 1k tbn, 1k tbc (default)
Stream mapping:
  Stream #0:0 -> #0:0 (vp9 (native) -> h264 (h264_v4l2m2m))
Press [q] to stop, [?] for help
[h264_v4l2m2m @ 0x55caa7c4a0] driver 'bcm2835-codec' on card 'bcm2835-codec-isp'
[h264_v4l2m2m @ 0x55caa7c4a0] driver 'bcm2835-codec' on card 'bcm2835-codec-encode'
[h264_v4l2m2m @ 0x55caa7c4a0] Using device /dev/video11
[h264_v4l2m2m @ 0x55caa7c4a0] driver 'bcm2835-codec' on card 'bcm2835-codec-encode'
ff_v4l2_buffer_enqueue: VIDIOC_QBUF : fd 5, index 0 type 9
ff_v4l2_buffer_enqueue: VIDIOC_QBUF : fd 5, index 1 type 9
ff_v4l2_buffer_enqueue: VIDIOC_QBUF : fd 5, index 2 type 9
ff_v4l2_buffer_enqueue: VIDIOC_QBUF : fd 5, index 3 type 9
[h264_v4l2m2m @ 0x55caa7c4a0] Failed to set number of B-frames
    Last message repeated 1 times
[h264_v4l2m2m @ 0x55caa7c4a0] Failed to set header mode
[h264_v4l2m2m @ 0x55caa7c4a0] Failed to set gop size
[h264_v4l2m2m @ 0x55caa7c4a0] h264 profile not found
[h264_v4l2m2m @ 0x55caa7c4a0] Encoder adjusted: qmin (0), qmax (51)
[h264_v4l2m2m @ 0x55caa7c4a0] Failed to set minimum video quantizer scale
[h264_v4l2m2m @ 0x55caa7c4a0] Failed to set maximum video quantizer scale
Output #0, mov, to 'grb_2-v4l2.mov':
  Metadata:
    encoder         : Lavf58.29.100
    Stream #0:0: Video: h264 (h264_v4l2m2m) (avc1 / 0x31637661), yuv420p, 720x480 [SAR 8:9 DAR 4:3], q=2-31, 200 kb/s, 29.97 fps, 30k tbn, 29.97 tbc (default)
    Metadata:
      encoder         : Lavc58.54.100 h264_v4l2m2m
ff_v4l2_buffer_enqueue: VIDIOC_QBUF : fd 5, index 0 type 10
    Last message repeated 2 times
ff_v4l2_buffer_enqueue: VIDIOC_QBUF : fd 5, index 0 type 9
# repeat for a few pages
ff_v4l2_buffer_enqueue: VIDIOC_QBUF : fd 5, index 0 type 9
[mov @ 0x55caa7b350] Non-monotonous DTS in output stream 0:0; previous: 834834, current: 0; changing to 834835. This may result in incorrect timestamps in the output file.
frame=  835 fps=305 q=-0.0 Lsize=     545kB time=00:00:27.82 bitrate= 160.3kbits/s speed=10.2x
video:540kB audio:0kB subtitle:0kB other streams:0kB global headers:0kB muxing overhead: 0.762498%

real    0m2.823s
user    0m3.942s
sys     0m0.534s

```

At this point I'm kind of stuck.

Is there any way I can at least figure out if the problem is something system-wide or with the ffmpeg in the Jellyfin image?

billyzs, 2020-09-21:
@piaste what's the output of `v4l2-ctl -d 10 --list-formats-out` and `lsmod | grep v4l`?

piaste, 2020-10-15:
@billyzs 

`v4l2-ctl -d 10 --list-formats-out`: 

```
ioctl: VIDIOC_ENUM_FMT
        Type: Video Output Multiplanar

        [0]: 'H264' (H.264, compressed)
        [1]: 'MJPG' (Motion-JPEG, compressed)
        [2]: 'VP80' (VP8, compressed)
```

`lsmod | grep v4l`: 

```
bcm2835_v4l2           45056  0
bcm2835_mmal_vchiq     32768  3 bcm2835_codec,bcm2835_v4l2,bcm2835_isp
v4l2_mem2mem           32768  1 bcm2835_codec
videobuf2_vmalloc      20480  1 bcm2835_v4l2
videobuf2_v4l2         28672  4 bcm2835_codec,bcm2835_v4l2,v4l2_mem2mem,bcm2835_isp
videobuf2_common       53248  5 bcm2835_codec,videobuf2_v4l2,bcm2835_v4l2,v4l2_mem2mem,bcm2835_isp
videodev              262144  6 bcm2835_codec,videobuf2_v4l2,bcm2835_v4l2,videobuf2_common,v4l2_mem2mem,bcm2835_isp
mc                     49152  6 videodev,bcm2835_codec,videobuf2_v4l2,videobuf2_common,v4l2_mem2mem,bcm2835_isp
```