sakaki-/gentoo-on-rpi-64bit #188: conflict over openssl versions

tiborh, 2020-07-26:
There are two sets of programs I cannot install together due to some conflict over openssl:
1. net-misc/openssh, net-wireless/wpa_supplicant, dev-qt/tqtnetwork
2. net-vpn/tor

the default was set 1, but I now more need set 2, so I uninstalled the programs in set 1 to be able to install set 2. However, in the future I may need openssh so I was wondering if this conflict can be resolve somehow. 

I have already tried the --oneshot and --backtrack=30 options offered in various places, but none of them worked.

Some other facts that may  be helpful. 
1. After first install from image written to SD Card, 
2. a genup was executed and it finished successfully.

<hr />

an example run:
$ sudo emerge --verbose net-misc/openssh

Local copy of remote index is up-to-date and will be used.

Local copy of remote index is up-to-date and will be used.

These are the packages that would be merged, in order:

Calculating dependencies... done!
[binary   R    ] dev-libs/openssl-1.1.1g:0/1.1::gentoo  USE="asm bindist* tls-heartbeat zlib -rfc3779 -sctp -sslv3 -static-libs -test -vanilla" 4,656 KiB
[binary  N     ] net-misc/openssh-8.3_p1-r2::gentoo  USE="X bindist pam pie ssl -X509 (-audit) -debug -hpn -kerberos (-ldns) -libedit -libressl -livecd -sctp -security-key (-selinux) -static -test -xmss" 1,966 KiB

Total: 2 packages (1 new, 1 reinstall, 2 binaries), Size of downloads: 6,621 KiB

!!! Multiple package instances within a single package slot have been pulled
!!! into the dependency graph, resulting in a slot conflict:

dev-libs/openssl:0

  (dev-libs/openssl-1.1.1g:0/1.1::gentoo, installed) USE="asm tls-heartbeat zlib -bindist -rfc3779 -sctp -sslv3 -static-libs -test -vanilla" pulled in by
    dev-libs/openssl:0=[-bindist] required by (net-vpn/tor-0.4.4.1_alpha:0/0::gentoo, installed) USE="man seccomp server -caps -doc -libressl -lzma -scrypt (-selinux) -systemd -test -tor-hardening -zstd"
                        ^^^^^^^^                                                                                                                                                                                                                                                                                                                                                                                                                                                                      
    dev-libs/openssl:0/1.1=[-bindist] required by (net-vpn/tor-0.4.4.1_alpha:0/0::gentoo, installed) USE="man seccomp server -caps -doc -libressl -lzma -scrypt (-selinux) -systemd -test -tor-hardening -zstd"
                            ^^^^^^^^                                                                                                                                                                                                                                                                                                                                                                                                                                                                      

  (dev-libs/openssl-1.1.1g:0/1.1::gentoo, binary scheduled for merge) USE="asm bindist tls-heartbeat zlib -rfc3779 -sctp -sslv3 -static-libs -test -vanilla" pulled in by
    >=dev-libs/openssl-1.1.0g:0[bindist] required by (net-misc/openssh-8.3_p1-r2:0/0::gentoo, binary scheduled for merge) USE="X bindist pam pie ssl -X509 (-audit) -debug -hpn -kerberos (-ldns) -libedit -libressl -livecd -sctp -security-key (-selinux) -static -test -xmss"
                                ^^^^^^^                                                                   
<hr />

Please let me know what other data is needed for successful investigation. I'm a first time user, so not experienced with Gentoo at all. (Previous experience extends to Raspbian, Ubuntu, and Manjaro only)

tiborh, 2020-08-01:
example of a run that genup recommends:
(based on the data provided, do you think it is safe to suppose that this is an unresolvable conflict?)

> $ sudo emerge --update --newuse --deep --with-bdeps=y --backtrack=30 @world
> 
> Local copy of remote index is up-to-date and will be used.
> 
> Local copy of remote index is up-to-date and will be used.
> Calculating dependencies... done!
> [binary   R    ] dev-libs/liblinear-230  USE="(-blas%)" 
> [binary   R    ] net-libs/libnatpmp-20150609  USE="(-static-libs%)" 
> [binary   R    ] dev-libs/json-c-0.14-r3  USE="-cpu-flags-x86-rdrand%" 
> [binary   R    ] media-libs/flac-1.3.3  CPU_FLAGS_PPC="-altivec% -vsx%" 
> [binary   R    ] dev-lang/orc-0.4.31  USE="-gtk-doc%" 
> [binary   R    ] dev-libs/openssl-1.1.1g  USE="bindist*" 
> [binary   R    ] net-misc/iperf-3.8.1  USE="(-profiling%)" 
> [binary   R    ] dev-python/enum34-1.1.6-r1  PYTHON_TARGETS="(-pypy%)" 
> [binary   R    ] dev-python/typing-3.7.4.1  PYTHON_TARGETS="(-pypy%)" 
> [binary  N     ] dev-perl/File-ShareDir-Install-0.110.0  USE="-test" 
> [binary   R    ] media-libs/libid3tag-0.15.1b-r4  USE="(-static-libs%)" 
> [binary   R    ] dev-libs/newt-0.52.21-r1  PYTHON_TARGETS="-python3_9%" 
> [binary   R    ] media-libs/libraw-0.19.5  USE="(-jpeg2k%)" 
> [binary  N     ] dev-qt/qtnetwork-5.15.0  USE="bindist ssl -connman -debug -gssapi (-libproxy) -libressl -networkmanager -sctp -test" 
> [binary   R    ] media-libs/netpbm-10.76.00  USE="(-jpeg2k%)" 
> [binary   R    ] media-gfx/dcraw-9.28.0-r1  USE="(-jpeg2k%)" 
> [binary   R    ] dev-python/markupsafe-1.1.1  PYTHON_TARGETS="-python3_9%" 
> [binary   R    ] dev-python/pygments-2.5.2  PYTHON_TARGETS="-python3_9%" 
> [binary   R    ] dev-python/extras-1.0.0  PYTHON_TARGETS="-python3_9%" 
> [binary   R    ] dev-python/ply-3.11  PYTHON_TARGETS="-python3_9%" 
> [binary   R    ] dev-python/chardet-3.0.4  PYTHON_TARGETS="-python3_9%" 
> [binary   R    ] dev-python/PySocks-1.7.1  PYTHON_TARGETS="-python3_9%" 
> [binary   R    ] dev-python/olefile-0.46  PYTHON_TARGETS="-python3_9%" 
> [binary   R    ] dev-python/distro-1.5.0  PYTHON_TARGETS="-python3_9%" 
> [binary   R    ] dev-python/docutils-0.16  PYTHON_TARGETS="-python3_9%" 
> [binary   R    ] dev-python/linecache2-1.0.0  PYTHON_TARGETS="-python3_9%" 
> [binary   R    ] dev-python/pyrsistent-0.15.6  PYTHON_TARGETS="-python3_9%" 
> [binary   R    ] dev-python/pillow-6.2.2  PYTHON_TARGETS="-python3_9%" 
> [binary   R    ] dev-python/traceback2-1.4.0  USE="-test%" PYTHON_TARGETS="-python3_9%" 
> [binary   R    ] dev-python/unittest2-1.1.0  USE="-test%" PYTHON_TARGETS="-python3_9%" 
> [binary   R    ] dev-python/pyopenssl-19.1.0  PYTHON_TARGETS="-python3_9%" 
> [binary   R    ] dev-python/dbus-python-1.2.16  PYTHON_TARGETS="-python3_9%" 
> [binary  N     ] net-misc/openssh-8.3_p1-r2  USE="X bindist pam pie ssl -X509 (-audit) -debug -hpn -kerberos (-ldns) -libedit -libressl -livecd -sctp -security-key (-selinux) -static -test -xmss" 
> [binary   R    ] dev-python/pycairo-1.18.2  PYTHON_TARGETS="-python3_9%" 
> [binary   R    ] dev-python/pygobject-3.34.0  PYTHON_TARGETS="-python3_9%" 
> [binary     U  ] xfce-base/libxfce4ui-4.15.3 [4.14.1] USE="-libgtop%" 
> [binary     U  ] xfce-base/exo-4.15.1 [0.12.11]
> [binary   R    ] xfce-base/garcon-0.7.0  USE="-gtk2*" 
> [binary     U  ] xfce-base/xfce4-panel-4.15.3 [4.14.4]
> [binary  N     ] net-wireless/wpa_supplicant-2.9-r2  USE="bindist dbus fils hs2-0 mbo mesh qt5 readline -ap -broadcom-sta -eap-sim -eapol-test -fasteap -libressl -macsec -p2p -privsep (-ps3) (-selinux) -smartcard -tdls -uncommon-eap-types (-wimax) -wps" 
> [binary   R    ] app-crypt/gpgme-1.13.1  PYTHON_TARGETS="-python3_8%" 
> [binary     U  ] xfce-base/xfce4-settings-4.15.1-r1 [4.14.3]
> [binary   R    ] dev-python/fixtures-3.0.0  PYTHON_TARGETS="-python3_9%" 
> [binary     U  ] xfce-base/thunar-4.15.1 [1.8.15]
> [binary     U  ] dev-embedded/rpi-64bit-meta-1.6.0-r5 [1.6.0-r4]
> 
> !!! Multiple package instances within a single package slot have been pulled
> !!! into the dependency graph, resulting in a slot conflict:
> 
> dev-libs/openssl:0
> 
>   (dev-libs/openssl-1.1.1g:0/1.1::gentoo, binary scheduled for merge) USE="asm bindist tls-heartbeat zlib -rfc3779 -sctp -sslv3 -static-libs -test -vanilla" pulled in by
>     >=dev-libs/openssl-1.1.0g:0[bindist] required by (net-misc/openssh-8.3_p1-r2:0/0::gentoo, binary scheduled for merge) USE="X bindist pam pie ssl -X509 (-audit) -debug -hpn -kerberos (-ldns) -libedit -libressl -livecd -sctp -security-key (-selinux) -static -test -xmss"
>                                 ^^^^^^^                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                         
>     >=dev-libs/openssl-1.0.2k:0/1.1=[bindist] required by (net-wireless/wpa_supplicant-2.9-r2:0/0::gentoo, binary scheduled for merge) USE="bindist dbus fils hs2-0 mbo mesh qt5 readline -ap -broadcom-sta -eap-sim -eapol-test - [... cut ...]
> 
> 
> It may be possible to solve this problem by using package.mask to
> prevent one of those packages from being selected. However, it is also
> possible that conflicting dependencies exist such that they are
> impossible to satisfy simultaneously.  If such a conflict exists in
> the dependencies of two different packages, then those packages can
> not be installed simultaneously.
> 
> For more information, see MASKED PACKAGES section in the emerge man
> page or refer to the Gentoo Handbook.

tiborh, 2020-08-01:
and the original genup run, based of which the above was executed:

> $ sudo genup
```

> * Checking Portage configuration, please wait...
> * Gentoo System Updater v1.0.27
> * Updating Portage tree and syncing the eix cache
> * (this may take some time)...
>  * Syncing all portage overlays
> /usr/bin/eix-sync: line 396: layman: command not found
>  * layman -S failed
>  * Running emerge --sync
> >>> Syncing repository 'gentoo' into '/var/db/repos/gentoo'...
>  * Using keys from /usr/share/openpgp-keys/gentoo-release.asc
>  * Refreshing keys via WKD ...                                           [ ok ]
> >>> Starting rsync with rsync://isshoni.org/gentoo-portage-pi64-gem...
> >>> Checking server timestamp ...
> receiving incremental file list
> timestamp.chk
> 
> Number of files: 1 (reg: 1)
> Number of created files: 0
> Number of deleted files: 0
> Number of regular files transferred: 1
> Total file size: 32 bytes
> Total transferred file size: 32 bytes
> Literal data: 32 bytes
> Matched data: 0 bytes
> File list size: 41
> File list generation time: 0.001 seconds
> File list transfer time: 0.000 seconds
> Total bytes sent: 104
> Total bytes received: 127
> 
> sent 104 bytes  received 127 bytes  154.00 bytes/sec
> total size is 32  speedup is 0.14
> 
> >>>
> >>> Timestamps on the server and in the local repository are the same.
> >>> Cancelling all further sync action. You are already up to date.
> >>>
> >>> In order to force sync, remove '/var/db/repos/gentoo/metadata/timestamp.chk'.
> >>>
> 
>  * Manifest timestamp: 2020-07-10 15:38:20 UTC
>  * Valid OpenPGP signature found:
>  * - primary key: DCD05B71EAB94199527F44ACDB6B8C1F96D8BF6D
>  * - subkey: E1D6ABB63BFCFB4BA02FDF1CEC590EEAC9189250
>  * - timestamp: 2020-07-10 15:38:20 UTC
> === Sync completed for gentoo
> >>> Syncing repository 'genpi64' into '/var/db/repos/genpi64'...
> >>> Syncing repository 'sakaki-tools' into '/var/db/repos/sakaki-tools'...
> /usr/bin/git fetch origin
> /usr/bin/git fetch origin
> Already up to date.
> === Sync completed for sakaki-tools
> remote: Enumerating objects: 78, done.
> remote: Counting objects: 100% (78/78), done.
> remote: Compressing objects: 100% (25/25), done.
> remote: Total 60 (delta 38), reused 52 (delta 33), pack-reused 0
> Unpacking objects: 100% (60/60), 62.24 KiB | 518.00 KiB/s, done.
> From https://github.com/sakaki-/genpi64-overlay
>    16e766b..a117c65  master     -> origin/master
> Updating 16e766b..a117c65
> Fast-forward
>  ....0-r4.ebuild => rpi-64bit-meta-1.6.0-r5.ebuild} |   1 -
>  .../package.accept_keywords/openscenegraph-openmw  |   1 +
>  .../rpi3/package.accept_keywords/recastnavigation  |   1 +
>  profiles/targets/rpi3/package.mask/exo             |   2 -
[... 213 lines cut ...]
>  *     gnome-base/gnome-keyring-3.34.0
>  * 
>  *   ~dev-qt/qtnetwork-5.15.0 pulled in by:
>  *     dev-qt/qthelp-5.15.0
>  * 
>  *   >=net-wireless/wpa_supplicant-2.9-r2 pulled in by:
>  *     dev-embedded/rpi-64bit-meta-1.6.0-r4
>  * 
>  * Have you forgotten to do a complete update prior to depclean? The
>  * most comprehensive command for this purpose is as follows:
>  * 
>  *   emerge --update --newuse --deep --with-bdeps=y @world
>  * 
>  * Note that the --with-bdeps=y option is not required in many
>  * situations. Refer to the emerge manual page (run `man emerge`)
>  * for more information about --with-bdeps.
>  * 
>  * Also, note that it may be necessary to manually uninstall
>  * packages that no longer exist in the repository, since it may not
>  * be possible to satisfy their dependencies.
> 
> * genup: Error: Caught signal - exiting
> 

```

AlexMikhalev, 2020-08-05:
I fixed openssl issue by adding `USE=-bindist`  into /etc/portage/make.conf (I also have ACCEPT_KEYWORDS="* ~*" there), 
and running 
```
sudo emerge --update --changed-use --deep --quiet @world
``` 
I needed both to complete glusterfs build dependencies.

tiborh, 2020-09-05:
Yes, I can confirm that. "Use=-bindist" was missing from my file. (sill no success with 'emerge', but they are different issues. and I am still figuring out if I can resolve the problems)

closed 2020-09-05 by tiborh