sakaki-/gentoo-on-rpi-64bit #191: www-apache/mod_perl will not build

TheMetMan, 2020-08-03:
I have a new install of the Headless Server on a Pi4 with 8Gb Memory.
I have got Apache Web Server running but need mod_perl.
I have added the accept.keywords for this:
```
>=www-apache/mod_perl-2.0.11 * ~*
>=dev-perl/Apache-Reload-0.130.0-r1 * ~*
>=dev-perl/Apache-SizeLimit-0.970.0-r2 * ~*

```
Then:
```
emerge -avqt www-apache/mod_perl

Local copy of remote index is up-to-date and will be used.
[nomerge      ] www-apache/mod_perl-2.0.11  USE="-debug -ithreads -test" 
[ebuild  N    ]  dev-perl/Apache-Reload-0.130.0-r1 
[ebuild  N    ]  dev-perl/Apache-SizeLimit-0.970.0-r2  USE="-test" 
[ebuild  N    ]   www-apache/mod_perl-2.0.11  USE="-debug -ithreads -test" 

```
and it fails with this error:
```
aarch64-unknown-linux-gnu-gcc -c -I/var/tmp/portage/www-apache/mod_perl-2.0.11/work/mod_perl-2.0.11/src/modules/perl -I/var/tmp/portage/www-apache/mod_perl-2.0.11/work/mod_perl-2.0.11/xs -I/usr/include/apr-1 -I/usr/include/apr-1 -I/usr/include/db6.0 -I/usr/include/apache2 -I/var/tmp/portage/www-apache/mod_perl-2.0.11/work/mod_perl-2.0.11/src/modules/perl -I/var/tmp/portage/www-apache/mod_perl-2.0.11/work/mod_perl-2.0.11/xs -I/usr/include/apr-1 -I/usr/include/apr-1 -I/usr/include/db6.0 -I/usr/include/apache2 -I/var/tmp/portage/www-apache/mod_perl-2.0.11/work/mod_perl-2.0.11/src/modules/perl -I/var/tmp/portage/www-apache/mod_perl-2.0.11/work/mod_perl-2.0.11/xs -I/usr/include/apr-1 -I/usr/include/apr-1 -I/usr/include/db6.0 -I/usr/include/apache2  -I/var/tmp/portage/www-apache/mod_perl-2.0.11/work/mod_perl-2.0.11/src/modules/perl -I/var/tmp/portage/www-apache/mod_perl-2.0.11/work/mod_perl-2.0.11/xs -I/usr/include/apr-1 -I/usr/include/apr-1 -I/usr/include/db6.0 -I/usr/include/apache2 -fwrapv -fno-strict-aliasing -pipe -D_LARGEFILE_SOURCE -D_FILE_OFFSET_BITS=64 -DMOD_PERL -DMP_COMPAT_1X -fgnu89-inline -DLINUX -D_REENTRANT -D_GNU_SOURCE  -march=armv8-a+crc -mtune=cortex-a72 -ftree-vectorize -O2 -pipe   -DVERSION=\"2.000011\" -DXS_VERSION=\"2.000011\" -fPIC "-I/usr/lib64/perl5/5.30.3/aarch64-linux/CORE" -DMP_HAVE_APR_LIBS -DMP_HAVE_APR_LIBS -DMP_HAVE_APR_LIBS  -DMP_HAVE_APR_LIBS Util.c
rm -f ../../../blib/arch/auto/ModPerl/Util/Util.so
aarch64-unknown-linux-gnu-gcc  -shared -march=armv8-a+crc -mtune=cortex-a72 -ftree-vectorize -O2 -pipe -Wl,-O1 -Wl,--as-needed  Util.o -Wl,-O1 -Wl,--as-needed -o ../../../blib/arch/auto/ModPerl/Util/Util.so  \
      \
  
chmod 755 ../../../blib/arch/auto/ModPerl/Util/Util.so
make[3]: Leaving directory '/var/tmp/portage/www-apache/mod_perl-2.0.11/work/mod_perl-2.0.11/WrapXS/ModPerl/Util'
make[2]: Leaving directory '/var/tmp/portage/www-apache/mod_perl-2.0.11/work/mod_perl-2.0.11/WrapXS/ModPerl'
make[1]: Leaving directory '/var/tmp/portage/www-apache/mod_perl-2.0.11/work/mod_perl-2.0.11/WrapXS'
 * ERROR: www-apache/mod_perl-2.0.11::gentoo failed (compile phase):
 *   emake failed
```
I cannot see anything here to give me a clue as to the cause.
So output of:
```
emerge --info '=www-apache/mod_perl-2.0.11::gentoo'
Portage 2.3.103 (python 3.6.10-final-0, default/linux/arm64/17.0/desktop, gcc-9.3.0, glibc-2.31-r5, 5.4.45-v8-3b41649ff96d-p4-bis+ aarch64)
=================================================================
                         System Settings
=================================================================
System uname: Linux-5.4.45-v8-3b41649ff96d-p4-bis+-aarch64-with-gentoo-2.7
KiB Mem:     7949164 total,   7203232 free
KiB Swap:    8191996 total,   8191996 free
Timestamp of repository gentoo: Fri, 10 Jul 2020 16:00:01 +0000
Head commit of repository gentoo: 99596149e4d2382e870a71c49c39f8e115c126f3
Head commit of repository sakaki-tools: ac4852fe85b8bd495c956d18c812c4eb9c65a2eb

Head commit of repository genpi64: a117c650d4e10824e9551c8e6a9f3f35c01bc0fe

sh bash 5.0_p17
ld GNU ld (Gentoo 2.33.1 p2) 2.33.1
distcc 3.3.3 aarch64-unknown-linux-gnu [disabled]
app-shells/bash:          5.0_p17::gentoo
dev-lang/perl:            5.30.3-r1::gentoo
dev-lang/python:          2.7.18::gentoo, 3.6.10-r2::genpi64, 3.7.8::genpi64, 3.8.3::genpi64, 3.9.0_beta4::gentoo
dev-util/cmake:           3.17.3::gentoo
dev-util/pkgconfig:       0.29.2::gentoo
sys-apps/baselayout:      2.7::gentoo
sys-apps/openrc:          0.42.1-r3::genpi64
sys-apps/sandbox:         2.20::gentoo
sys-devel/autoconf:       2.13-r1::gentoo, 2.69-r5::gentoo
sys-devel/automake:       1.16.2::gentoo
sys-devel/binutils:       2.33.1-r1::gentoo
sys-devel/gcc:            9.3.0::gentoo
sys-devel/gcc-config:     2.3.1::gentoo
sys-devel/libtool:        2.4.6-r6::gentoo
sys-devel/make:           4.3::gentoo
sys-kernel/linux-headers: 5.4-r1::gentoo (virtual/os-headers)
[... 28 lines cut ...]
ACCEPT_LICENSE="@FREE CC-Sampling-Plus-1.0"
CBUILD="aarch64-unknown-linux-gnu"
CFLAGS="-march=armv8-a+crc -mtune=cortex-a72 -ftree-vectorize -O2 -pipe"
CHOST="aarch64-unknown-linux-gnu"
CONFIG_PROTECT="/boot/cmdline.txt /boot/config.txt /boot/startup.sh /etc /usr/share/gnupg/qualified.txt"
CONFIG_PROTECT_MASK="/etc/ca-certificates.conf /etc/dconf /etc/env.d /etc/fonts/fonts.conf /etc/gconf /etc/gentoo-release /etc/revdep-rebuild /etc/sandbox.d /etc/terminfo"
CXXFLAGS="-march=armv8-a+crc -mtune=cortex-a72 -ftree-vectorize -O2 -pipe"
DISTDIR="/var/cache/distfiles"
EMERGE_DEFAULT_OPTS="--jobs=5 --load-average=4"
ENV_UNSET="DBUS_SESSION_BUS_ADDRESS DISPLAY GOBIN GOPATH PERL5LIB PERL5OPT PERLPREFIX PERL_CORE PERL_MB_OPT PERL_MM_OPT XAUTHORITY XDG_CACHE_HOME XDG_CONFIG_HOME XDG_DATA_HOME XDG_RUNTIME_DIR"
FCFLAGS="-O2"
FEATURES="assume-digests binpkg-docompress binpkg-dostrip binpkg-logs buildpkg config-protect-if-modified distlocks ebuild-locks fixlafiles getbinpkg ipc-sandbox merge-sync multilib-strict network-sandbox news parallel-fetch preserve-libs protect-owned qa-unresolved-soname-deps sandbox sfperms strict unknown-features-warn unmerge-logs unmerge-orphans userfetch userpriv usersandbox usersync xattr"
FFLAGS="-O2"
GENTOO_MIRRORS="http://gentoo.osuosl.org/ http://trumpetti.atm.tut.fi/gentoo/ http://distfiles.gentoo.org"
LANG="en_GB.utf8"
LDFLAGS="-Wl,-O1 -Wl,--as-needed"
MAKEOPTS="-j5 -l4"
PKGDIR="/var/cache/binpkgs"
PORTAGE_BINHOST="https://isshoni.org/pi64pie"
PORTAGE_CONFIGROOT="/"
PORTAGE_RSYNC_OPTS="--recursive --links --safe-links --perms --times --omit-dir-times --compress --force --whole-file --delete --stats --human-readable --timeout=180 --exclude=/distfiles --exclude=/local --exclude=/packages --exclude=/.git"
PORTAGE_TMPDIR="/var/tmp"
USE="X a52 aac acl acpi alsa arm64 berkdb bindist bluetooth branding bzip2 cairo cdda cdr cli crypt cups dbus dri dts dvdr egl elogind emboss encode exif ffmpeg flac fortran gdbm gif gles1 gles2 gpm gtk iconv icu ipv6 jpeg lcms ldap libnotify libtirpc lock mad mng mp3 mp4 mpeg ncurses nls nptl ogg opengl openmp pam pango pcre pdf png policykit ppds pulseaudio qt4 qt5 readline sdl seccomp spell split-usr ssl startup-notification svg tcpd thunar tiff truetype udev udisks unicode upower usb vorbis wxwidgets x264 xattr xcb xml xv xvid zlib" ADA_TARGET="gnat_2018" APACHE2_MODULES="authn_core authz_core socache_shmcb unixd actions alias auth_basic authn_alias authn_anon authn_dbm authn_default authn_file authz_dbm authz_default authz_groupfile authz_host authz_owner authz_user autoindex cache cgi cgid dav dav_fs dav_lock deflate dir disk_cache env expires ext_filter file_cache filter headers include info log_config logio mem_cache mime mime_magic negotiation rewrite setenvif speling status unique_id userdir usertrack vhost_alias" CALLIGRA_FEATURES="karbon sheets words" COLLECTD_PLUGINS="df interface irq load memory rrdtool swap syslog" CPU_FLAGS_ARM="edsp thumb thumb2 v4 v5 v6 v7 v8 vfp vfp-d32 vfpv3 vfpv4" ELIBC="glibc" GPSD_PROTOCOLS="ashtech aivdm earthmate evermore fv18 garmin garmintxt gpsclock greis isync itrax mtk3301 nmea ntrip navcom oceanserver oldstyle oncore rtcm104v2 rtcm104v3 sirf skytraq superstar2 timing tsip tripmate tnt ublox ubx" INPUT_DEVICES="evdev synaptics" KERNEL="linux" L10N="en en-GB" LCD_DEVICES="bayrad cfontz cfontz633 glk hd44780 lb216 lcdm001 mtxorb ncurses text" LIBREOFFICE_EXTENSIONS="presenter-console presenter-minimizer" LLVM_TARGETS="WebAssembly BPF" OFFICE_IMPLEMENTATION="libreoffice" PHP_TARGETS="php7-2" POSTGRES_TARGETS="postgres10 postgres11" PYTHON_SINGLE_TARGET="python3_7" PYTHON_TARGETS="python2_7 python3_5 python3_6 python3_7" QEMU_SOFTMMU_TARGETS="aarch64 arm i386 x86_64" QEMU_USER_TARGETS="aarch64" RUBY_TARGETS="ruby25 ruby27" USERLAND="GNU" VIDEO_CARDS="fbdev vc4" XTABLES_ADDONS="quota2 psd pknock lscan length2 ipv4options ipset ipp2p iface geoip fuzzy condition tee tarpit sysrq steal rawnat logmark ipmark dhcpmac delude chaos account"
Unset:  CC, CPPFLAGS, CTARGET, CXX, INSTALL_MASK, LC_ALL, LINGUAS, PORTAGE_BUNZIP2_COMMAND, PORTAGE_COMPRESS, PORTAGE_COMPRESS_FLAGS, PORTAGE_RSYNC_EXTRA_OPTS
```

sakaki-, 2020-10-30:
30 Oct 2020: sadly, due legal obligations arising from a recent change in my 'real world' job, I must announce I am **standing down as maintainer of this project with immediate effect**. For the meantime, I will leave the repo up (for historical interest, and since the images may be of use still in certain applications); however, there will be no further updates to the underlying binhost etc., nor will I be accepting / actioning further pull requests or bug reports from this point. Email requests for support will also have to be politely declined, so, **please treat this as an effective EOL notice**.<br><br>For further details, please see my post [here](https://www.raspberrypi.org/forums/viewtopic.php?p=1750206#p1750206).<br><br>Many thanks for your interest in this project!<br><br>With sincere apologies, sakaki ><

TheMetMan, 2020-10-30:
Such a shame @sakaki- that you are 'retiring'! 
As a Gentoo User for pretty much the whole of Gentoo's lifetime, I really enjoyed your project which I thought was VERY well put together. A lot of thought and expertise, from which I have benefited.
All the best for the future
Regards