sakaki-/gentoo-on-rpi-64bit #213: Cant update the system using genup

sindibad363, 2021-07-31:
After flashing the sd-card and ssh-ing into it i used commnad genup as root but it errored out in emerge --sync

<details><summary>Here is terminal output</summary>

```
* Checking Portage configuration, please wait...
* Gentoo System Updater v1.0.27
* Updating Portage tree and syncing the eix cache
* (this may take some time)...
 * Syncing all portage overlays
/usr/bin/eix-sync: line 396: layman: command not found
 * layman -S failed
 * Running emerge --sync
>>> Syncing repository 'gentoo' into '/var/db/repos/gentoo'...
 * Using keys from /usr/share/openpgp-keys/gentoo-release.asc
 * Refreshing keys via WKD ...                                                                                                                                                                                                                                            [ ok ]
>>> Starting rsync with rsync://isshoni.org/gentoo-portage-pi64-gem...
>>> Checking server timestamp ...
receiving incremental file list
timestamp.chk

Number of files: 1 (reg: 1)
Number of created files: 0
Number of deleted files: 0
Number of regular files transferred: 1
Total file size: 32 bytes
Total transferred file size: 32 bytes
Literal data: 32 bytes
Matched data: 0 bytes
File list size: 41
File list generation time: 0.001 seconds
File list transfer time: 0.000 seconds
Total bytes sent: 104
Total bytes received: 127

sent 104 bytes  received 127 bytes  154.00 bytes/sec
total size is 32  speedup is 0.14

>>>
>>> SERVER OUT OF DATE: rsync://isshoni.org/gentoo-portage-pi64-gem
>>>
>>> In order to force sync, remove '/var/db/repos/gentoo/metadata/timestamp.chk'.
>>>

>>> Retrying...
!!! Exhausted addresses for isshoni.org
>>> Syncing repository 'genpi64' into '/var/db/repos/genpi64'...
>>> Syncing repository 'sakaki-tools' into '/var/db/repos/sakaki-tools'...
/usr/bin/git fetch origin
/usr/bin/git fetch origin
Already up to date.
=== Sync completed for genpi64
Already up to date.
=== Sync completed for sakaki-tools

 * IMPORTANT: 1 news items need reading for repository 'genpi64'.
 * IMPORTANT: 1 news items need reading for repository 'sakaki-tools'.
 * IMPORTANT: 8 news items need reading for repository 'gentoo'.
 * Use eselect news read to view new items.

Action: sync for repo: gentoo, returned code = 1
Action: sync for repo: genpi64, returned code = 0
Action: sync for repo: sakaki-tools, returned code = 0

 * An update to portage is available. It is _highly_ recommended
 * that you update portage now, before any other packages are updated.

 * To update portage, run 'emerge --oneshot sys-apps/portage' now.

 * emerge --sync failed
 * Time statistics:
     7 seconds for syncing
     7 seconds total

* genup: Error: Caught signal - exiting

```

</details>

If you have any suggestions how can I upgrade packages please hit me up. (Also kinda new to gentoo, too)

duy-nguyen-com, 2021-11-13:
when updating all your packages, usually you do a @world set update, however I think its just that sakaki-tools and genpi64 repos are down

emerge -uDN @world

duy-nguyen-com, 2021-11-13:
"Keeping Your System Up-To-Date

You can update your system at any time. As there are quite a few steps involved to do this correctly on Gentoo, I have provided a convenience script, genup (source, manpage) to do this as part of the image.

Note that by default, this script will run automatically once per week, commencing the second week after first boot (see the task installed in /etc/cron.weekly). Review the logfile /var/log/latest-genup-run.log to see changes made by the latest auto-update run."

> try doing genup